espressif/esp-adf #441: Error compiling v2.1

szucsi23, 2020-08-06:
Hey!
While building our project, which uses adf v2.1, I get the following error messages:

`C:/esp32/esp-adf/components/audio_stream/http_stream.c: In function 'get_audio_type':
C:/esp32/esp-adf/components/audio_stream/http_stream.c:95:16: error: 'ESP_AUDIO_TYPE_M3U8' undeclared (first use in this function)
         return ESP_AUDIO_TYPE_M3U8;
                ^
C:/esp32/esp-adf/components/audio_stream/http_stream.c:95:16: note: each undeclared identifier is reported only once for each function it appears in
C:/esp32/esp-adf/components/audio_stream/http_stream.c:99:16: error: 'ESP_AUDIO_TYPE_PLS' undeclared (first use in this function)        
         return ESP_AUDIO_TYPE_PLS;
                ^
C:/esp32/esp-adf/components/audio_stream/http_stream.c: In function '_is_playlist':
C:/esp32/esp-adf/components/audio_stream/http_stream.c:142:28: error: 'ESP_AUDIO_TYPE_M3U8' undeclared (first use in this function)
     if (info->codec_fmt == ESP_AUDIO_TYPE_M3U8 || info->codec_fmt == ESP_AUDIO_TYPE_PLS)
                            ^
C:/esp32/esp-adf/components/audio_stream/http_stream.c:142:70: error: 'ESP_AUDIO_TYPE_PLS' undeclared (first use in this function)
     if (info->codec_fmt == ESP_AUDIO_TYPE_M3U8 || info->codec_fmt == ESP_AUDIO_TYPE_PLS)
                                                                      ^
C:/esp32/esp-adf/components/audio_stream/http_stream.c: In function '_resolve_playlist':
CC build/bootloader/bootloader_support/src/flash_qio_mode.oC:/esp32/esp-adf/components/audio_stream/http_stream.c:268:27: error: 'ESP_AUDIO_TYPE_PLS' undeclared (first use in this function)
     if (info.codec_fmt == ESP_AUDIO_TYPE_PLS)
                           ^

make[1]: *** [/c/esp32/esp-idf/make/component_wrapper.mk:290: http_stream.o] Error 1
make: *** [C:/esp32/esp-idf/make/project.mk:552: component-audio_stream-build] Error 2`

And even searching on github for the missing defined, I can't find them.
Any guess what could cause the problem?
I pulled master, checked out v2.1, did a clean build, nothing solves it.
Thanks in advance!

jason-mao, 2020-08-07:
@szucsi23  The error due to missed the `esp-adf-libs`. If you download .zip file directly, the ADF submodule would not include.

szucsi23, 2020-08-12:
I cloned the repo from terminal, but yeah, "git submodule update --recursive" solved it, thanks!

closed 2020-08-12 by szucsi23