espressif/esp-adf #445: flexible pipline with Codec switching

marrog86, 2020-08-12:
Hello, 

Pleas help me with the correct way to switch the Codec in respect of the http header. 

Problem: 
Http thread hangs; onliy working with the most first  Codec

function for changin codec: 

```
audio_pipeline_breakup_elements(player->pipeline, player->currentCodec);
char *list[] =......
ESP_LOGW(TAG, "switching Decoder to %s", player->codecs[type].name);
audio_pipeline_relink(player->pipeline, list, 4);
audio_pipeline_set_listener(player->pipeline, player->event_if);	
```	

My situation:
1. setting up the pipeline for mp3 
2. run http stream
-> works

1. setting up the pipeline for mp3, 
2. relinking to wav
3. relinking to mp3
4. run http stream
-> works

1. setting up the pipeline for wav
2. relinking to mp3
3. run http stream
-> error:
```
I (2727) AUDIO_PIPELINE: audio_pipeline_breakup_elements IN,0x3f8014e8,wav
I (2727) AUDIO_PIPELINE: 767, el:3f801218, http, in_rb:00000000, out_rb:3f801918, linked:1, el-kept:0
W (2737) AUDIO_PIPELINE: found kept_ctx_el and ringbuf, but not set kept
I (2747) AUDIO_PIPELINE: 767, el:3f8013a0, mp3, in_rb:00000000, out_rb:00000000, linked:0, el-kept:0
I (2757) AUDIO_PIPELINE: 767, el:3f8014e8, wav, in_rb:3f801918, out_rb:3f901b14, linked:1, el-kept:0
W (2767) AUDIO_PIPELINE: found kept_ctx_el and ringbuf, but not set kept
I (2767) AUDIO_PIPELINE: 767, el:3f801640, equ, in_rb:3f901b14, out_rb:3f901b50, linked:1, el-kept:0
I (2777) AUDIO_PIPELINE: output ringbuf is null
I (2787) AUDIO_PIPELINE: 767, el:3f8017e0, i2s, in_rb:3f901b50, out_rb:00000000, linked:1, el-kept:0
I (2797) AUDIO_PIPELINE: output ringbuf is null
W (2797) PLAYER: switching Decoder to mp3
W (2807) AUDIO_PIPELINE: 925, linked:0, kept:0, el:http, el:0x3f801218, tag:http, target-el:0x3f801218
I (2957) AUDIO_PIPELINE: create new rb,rb:0x3f901bac
W (2957) AUDIO_PIPELINE: 925, linked:1, kept:0, el:http, el:0x3f801218, tag:mp3, target-el:0x3f8013a0
W (2967) AUDIO_PIPELINE: 925, linked:0, kept:0, el:mp3, el:0x3f8013a0, tag:mp3, target-el:0x3f8013a0
W (2977) AUDIO_PIPELINE: 925, linked:1, kept:0, el:http, el:0x3f801218, tag:equ, target-el:0x3f801640
W (2987) AUDIO_PIPELINE: 925, linked:1, kept:0, el:mp3, el:0x3f8013a0, tag:equ, target-el:0x3f801640
W (2997) AUDIO_PIPELINE: 925, linked:0, kept:0, el:wav, el:0x3f8014e8, tag:equ, target-el:0x3f801640
W (3007) AUDIO_PIPELINE: 925, linked:0, kept:0, el:equ, el:0x3f801640, tag:equ, target-el:0x3f801640
W (3017) AUDIO_PIPELINE: 925, linked:1, kept:0, el:http, el:0x3f801218, tag:i2s, target-el:0x3f8017e0
W (3027) AUDIO_PIPELINE: 925, linked:1, kept:0, el:mp3, el:0x3f8013a0, tag:i2s, target-el:0x3f8017e0
W (3037) AUDIO_PIPELINE: 925, linked:0, kept:0, el:wav, el:0x3f8014e8, tag:i2s, target-el:0x3f8017e0
W (3047) AUDIO_PIPELINE: 925, linked:1, kept:0, el:equ, el:0x3f801640, tag:i2s, target-el:0x3f8017e0
W (3057) AUDIO_PIPELINE: 925, linked:0, kept:0, el:i2s, el:0x3f8017e0, tag:i2s, target-el:0x3f8017e0
I (3067) AUDIO_THREAD: The http task allocate stack on external memory
I (3067) AUDIO_ELEMENT: [http-0x3f801218] Element task created
I (3077) AUDIO_THREAD: The mp3 task allocate stack on external memory
I (3097) AUDIO_ELEMENT: [mp3-0x3f8013a0] Element task created
I (3097) AUDIO_ELEMENT: [equ-0x3f801640] Element task created
I (3107) AUDIO_ELEMENT: [i2s-0x3f8017e0] Element task created
I (3107) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:364, MEM Total:1954120 Bytes, Inter:153644 Bytes, Dram:137640 Bytes

I (3127) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
I (3127) HTTP_STREAM: _http_open
I (3127) HTTP_STREAM: URI=https://oe3shoutcast.sf.apa.at/;
I (3137) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (3147) AUDIO_ELEMENT: [equ] AEL_MSG_CMD_RESUME,state:1
I (3147) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (3157) I2S_STREAM: AUDIO_STREAM_WRITER
I (3157) AUDIO_PIPELINE: Pipeline started
W (3167) PLAYER: Run
I (3167) PLAYER: Volume: 10 ; Track:  https://oe3shoutcast.sf.apa.at/; 
I (3177) WIFI: got ip:192.168.178.28
I (3177) WIFI: connected -> switching off  AP
I (3187) esp_netif_handlers: sta ip: 192.168.178.28, mask: 255.255.255.0, gw: 192.168.178.1
I (3147) MP3_DECODER: MP3 opened
W (4797) HTTP_STREAM: Content-Type = audio/mpeg
I (4797) HTTP_STREAM: total_bytes=0
E (9817) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (9817) task_wdt:  - IDLE0 (CPU 0)
E (9817) task_wdt: Tasks currently running:
E (9817) task_wdt: CPU 0: mp3
E (9817) task_wdt: CPU 1: IDLE1
I (12127) PLAYER: i2s_stream_writer -> AEL_STATUS_STATE_RUNNING 
I (12127) PLAYER: equalizer -> AEL_STATUS_STATE_RUNNING 
I (12127) PLAYER:  Receive music info from HTTP decoder, sample_rates=44100, bits=16, ch=2, length=0
I (12137) PLAYER: http_stream_reader -> AEL_STATUS_STATE_RUNNING 
I (12137) PLAYER: decoder -> AEL_STATUS_STATE_RUNNING 
I (12147) PLAYER:  Receive music info from CODEC decoder, sample_rates=48000, bits=16, ch=2, length=0
I (12177) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_PAUSE
I (12197) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
W (12207) AUDIO_ELEMENT: [i2s] RESUME timeout
I (12207) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:3
I (12217) I2S_STREAM: AUDIO_STREAM_WRITER
I (12217) PLAYER: i2s_stream_writer -> AEL_STATUS_STATE_PAUSED  
I (12237) PLAYER: i2s_stream_writer -> AEL_STATUS_STATE_RUNNING 
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4009c0fb  PS      : 0x00060d33  A0      : 0x8009b275  A1      : 0x3fa40170  
0x4009c0fb: uxListRemove at /home/martin/esp/41/esp-idf/components/freertos/list.c:213

A2      : 0x0000001d  A3      : 0x3ffddc58  A4      : 0x00060d23  A5      : 0x000000fe  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fa40150  
A10     : 0x00000001  A11     : 0x3ffddc58  A12     : 0x00060d23  A13     : 0x000000fe  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x40090164  LEND    : 0x40090192  LCOUNT  : 0x00000000  
0x40090164: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:168

0x40090192: memcpy at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:201

ELF file SHA256: 29135dce16a90ae0

Backtrace: 0x4009c0f8:0x3fa40170 |<-CORRUPTED
0x4009c0f8: uxListRemove at /home/martin/esp/41/esp-idf/components/freertos/list.c:210

Rebooting...

```

jason-mao, 2020-09-30:
@marrog86 How about your problem? As your description ,instead `audio_pipeline_unlink ` break the pipeline, then linked a new one.

marrog86, 2020-10-21:
It seems for me tat it is not possible to abort end reset he elements correctly. So y made a complete stop procedure (with wait until stop) like in the examples before starting new one. This takes a lot of time for Music which yet should not be played so i tried to avoid this. 

Would be great having an fast cancel y forget all buffer data Procedure for starting new Tracks immediatly without making the hole stopping procedure.

fjullien, 2021-08-03:
@marrog86 Did you find a good way to correctly to switch from one decoder to another at runtime ?

closed 2023-09-05 by jason-mao