espressif/esp-adf #448: 在micopython 里使用adf 播放 flash 根目录 mp3文件 报错

tongjinlv, 2020-08-26:
import time
import audio   
def callback(stat):
    pass
def wait_playing(p):
    i=1
    while p.get_state()['status'] == audio.player.STATUS_RUNNING:
        print('\rPlaying: %s' % ('*' * i), end='')
        time.sleep(1)
        i += 1
    print('\n')

mPlayer=audio.player(callback)
mPlayer.set_vol(40)
print('Set volume to: ', mPlayer.get_vol())
mPlayer.play('file://test.wav')
print("zzzzzzzzzzzzzzzzzz")
wait_playing(mPlayer)
print(audio.mem_info())
代码如上=============
错误如下=============

[0;33mW (12013) I2S: I2S driver already installed[0m
[0;32mI (12023) I2S: APLL: Req RATE: 16000, real rate: 15999.986, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 4095996.500, SCLK: 511999.562500, diva: 1, divb: 0[0m
Set volume to:  0
[0;32mI (12033) ESP_AUDIO_CTRL: Enter play procedure, src:0[0m
[0;33mW (12043) ESP_AUDIO_CTRL: This URI has no extension and decoder type is not 'auto'[0m
[0;32mI (12043) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5, que:0x3ffbd844[0m
[0;32mI (12053) ESP_AUDIO_CTRL: Func:_ctrl_play, Line:713, MEM Total:2944864 Bytes, Inter:191764 Bytes, Dram:174708 Bytes

[0m
[0;32mI (12053) ESP_AUDIO_TASK: It's a decoder[0m
[0;32mI (12073) ESP_AUDIO_TASK: 1.CUR IN:[IN_file],CODEC:[NULL],RESAMPLE:[48000],OUT:[OUT_iis],rate:0,ch:0,pos:0[0m
[0;32mI (12083) AUDIO_ELEMENT: [IN_file-0x3f954ae8] Element task created[0m
[0;32mI (12093) AUDIO_ELEMENT: [IN_file] AEL_MSG_CMD_RESUME,state:1[0m
[0;33mW (12093) ESP_AUDIO_TASK: Try to got music format by in stream[0m
[0;32mI (12113) VFS_STREAM: File size is 29078 byte,pos:0[0m
[0;32mI (12113) ESP_AUDIO_TASK: Get codec type occur with element status [0xc][0m
[0;32mI (12123) AUDIO_ELEMENT: OUT-[IN_file] AEL_IO_DONE,0[0m
[0;32mI (12123) ESP_AUDIO_TASK: 2.Handles,IN:0x3f954ae8,CODEC:0x0,FILTER:0x3f955830,OUT:0x3f955688[0m
[0;31mE (12133) ESP_AUDIO_TASK: NULL does not supported[0m
[0;32mI (12133) ESP_AUDIO_CTRL: Exit play procedure, ret:81003[0m
[0;32mzzzzzzzzzzzzzzzzzz
I (12133) E

S{'dram': 170720, 'inter': 187776, 'mem_total': 2938396}
>>> P_AUDIO_TASK: Func:media_ctrl_task, Line:641, MEM Total:2938396 Bytes, Inter:187776 Bytes, Dram:170720 Bytes

[0m

jason-mao, 2020-09-02:
@tongjinlv 你可以把你的播放文件位置改为mPlayer.play('file://music/test.wav')再测试一下吗？

liuyunrui123, 2020-11-06:
播放路径改为 file://sdcard/test.wav  进行测试

jason-mao, 2020-12-02:
> 播放路径改为 file://sdcard/test.wav 进行测试

@liuyunrui123 测试如何？

liuyunrui123, 2020-12-02:
我就是按这个路径来测试的，跟这个没关系，因为其他音乐文件都是一样播放的，我已经确定是代码内部的bug已经修复了

closed 2021-09-26 by jason-mao