espressif/esp-adf #450: switching from v2.0 to v2.1, problems in voip apps

szucsi23, 2020-08-28:
Hey!
We've also discovered that switching from v2.0-beta3 to v2.1 causes two issues in different VOIP apps.

Linphone:
-using v2.0-beta3 works fine, but when we switch to v2.1, we get a 488 error, when trying to call a user, who uses Linphone. (nothing changes i the code, except for the pipeline parts, that were needed to use the updated adf). Attached logs from both calls, below.

SessionTalk:
-In v2.0-beta3 the DTMF works fine, but after the update, we don't get the dtmf messages on the board (I've tried all three settings, incl.: RFC2833)

Do you have any ideas, what we could try to solve these issues? We've spent the past few days debugging with no luck.

[2v0b3-antisip-good.txt](https://github.com/espressif/esp-adf/files/5141915/2v0b3-antisip-good.txt)
[2v1-antisip-err488.txt](https://github.com/espressif/esp-adf/files/5141916/2v1-antisip-err488.txt)

szucsi23, 2020-09-01:
From the antlr error log it seems there is an issue in the CRLF (end of line) characters in the SDP:

```
2020-08-26 15:01:06:255 [org.linphone/belle-sip] MESSAGE [org.antlr.runtime.MismatchedTokenException] reason [()* loopback of 411:22: ( attribute CR LF )*] at line[10] position[0]
2020-08-26 15:01:06:255 [org.linphone/belle-sip] MESSAGE [org.antlr.runtime.MismatchedTokenException] reason [()* loopback of 411:22: ( attribute CR LF )*] at line[10] position[0]
2020-08-26 15:01:06:255 [org.linphone/belle-sip] ERROR session_description parser error for
```

ahhfzhang, 2020-09-01:
so that's why we got `488` error ?

szucsi23, 2020-09-01:
Yes, its for the 488 problem, sorry I forgot to add that.
We've asked Linphone about this problem, and this was their answer.

closed 2020-11-03 by szucsi23