espressif/esp-adf #455: VOIP sleep mode - lyraT or Audio Kit v2.2

Patsjemoe, 2020-09-01:
Hi,
Anyone familiar with implementing a sleep mode for battery operated LyraT or Audio kit v2.2, taking into account the need for maintaining (minimum-not sure how much exactly..to be tested) wifi for connection with Asterisk VOIP server, and which should be woken up when for example the call button is pressed.  My assumption would be to use modem sleep, but it is hard to find a good description of how to implement it.
any ideas ?
thanks and regards
Ludo

ahhfzhang, 2020-09-01:
Hi, 
refer to the wifi power_save example, I think you just need to configure it at the beginning of app.

Patsjemoe, 2020-09-01:
OK, found it under esp-idf

Patsjemoe, 2020-09-05:
Hi,
I was looking to implement this into the VOIP application, but ran into a number of errors. It is also not just adding to the main program, but also had to adapt the kconfig.projbuild file to enable selection of  the wifi listen interval and power save mode.
When looking to my basic program running in msys, I noticed following line : 
I (1975) wifi:AP's beacon interval = 102400 us, DTIM period = 1

1) Where is this setting done ? 
2) As I have added also a wifi listen interval in the kconfig.projbuild file, is this setting conflictual with the setting of the beason interval ?

thanks for your help to clarify.
regards
Ludo

closed 2020-09-07 by Patsjemoe