espressif/esp-adf #456: Can we use AMR encoder-decoder without using an SD card? (Error: " AMR_DECODER: Bad header ")

utkutpcgl, 2020-09-01:
I tried to build a simple pipeline in which I have the following structure:
[codec_chip]-->i2s_stream-->amr_encoder-->amr_decoder-->i2s_stream-->[codec_chip]

I do not use any external storage device for the encoded data or to feed the decoder. Yet, I have tried this with "ESP-ADF pipeline" and by combining "ESP-ADF elements" with ring-buffers. After running the pipeline or running-resuming the elements, I received the same error as follows : 

[image]

My end goal is to have to esp32 lyrat to transfer and receive real-time compressed sound data simultaneously with ESP-NOW. For this purpose I think the best compression method is AMR, however, I doubt that the pipeline will not work in real-time. Since all the available examples use SD Card to feed the AMR decoder or AMR encoder to store the data. 

Yet, is it possible to transfer the AMR encoded data with ESP-NOW and decode it in the other device simultaneously bidirectional in real-time?

jason-mao, 2020-09-02:
@utkutpcgl 100% yes.  I think you should set the AMR encoder configuration `contain_amrnb_header` true on `amrnb_encoder_init`

utkutpcgl, 2020-09-02:
Thanks for your immediate response @jason-mao.
I am using the stable version which does not have _contain_amrnb_header_ configuration in the struct _amrnb_encoder_cfg_t_. I noticed it is available in the latest documentation. Should I upgrade my version or do you think I have made an error while installing the stable version? 
For time being I will try the latest version until I receive an answer.

jason-mao, 2020-09-02:
@utkutpcgl Yes, the `contain_amrnb_header ` available on latest commit,  I think you can try it.

utkutpcgl, 2020-09-02:
I applied the changes you suggested and now it works fine! 
I upgraded esp-idf from 4.0.1 to 4.1.0 and reinstalled the esp-adf API. I do not know why the previous 4.0.1 version did not work with the amr codec (Error: " AMR_DECODER: Bad header "). 
Can it be a bug? 

Thanks!

jason-mao, 2020-09-08:
@utkutpcgl 

> I upgraded esp-idf from 4.0.1 to 4.1.0 and reinstalled the esp-adf API. I do not know why the previous adf and 4.0.1 idf version did not work with the amr codec (Error: " AMR_DECODER: Bad header ").
Can it be a bug?

Uhm, the [`contain_amrnb_header `](https://github.com/espressif/esp-adf-libs/commit/a1fc0543c754224c61d53a1debfc5a28f4c4c334) is a new feature update a few weeks ago

closed 2020-09-08 by jason-mao

utkutpcgl, 2020-09-28:
Hello again,
One point is not clear to me and might be helpful for debugging esp-adf. When I downgraded esp-adf to 2.1 or 2.0 I received again the bad header error. 
I think there should not be any need for nb or wb header configuration in previous versions since it is not even available. So, why is there still the "AMR_DECODER: Bad header" exception. Is it related to esp-idf version? If there is a different idf version requirement for each adf version, then it should stated clearly. 
You said that contain_amrnb_header is a new facility, though, why does not the previous versions work without it?