espressif/esp-adf #457: mPlayer.play('file://sdcard/music/piano/36[49].mp3',0,False)          assert failed!

tongjinlv, 2020-09-04:
[0;32mI (32342) ESP_AUDIO_CTRL: Exit play procedure, ret:0[0m
[0;32mI (32552) ESP_AUDIO_CTRL: Enter media_ctrl_stop procedure, src:103, type:0, st:RUNNING, phase:0[0m
[0;32mI (32552) ESP_AUDIO_CTRL: media_ctrl_stop, src:103, type:0, st:RUNNING, phase:0, is_sent_play_cmd:0[0m
[0;32mI (32562) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:2, Available:5, que:0x3ffdd7cc[0m
[0;32mI (32572) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_STOP, status:RUNNING[0m
/home/work/PRJ/mata_esp32/submodule/esp-idf/components/freertos/queue.c:729 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40094cf0 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40097d78:0x3f901b60 |<-CORRUPTED

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5224
load:0x40078000,len:14040
load:0x40080400,len:3764
entry 0x40080630
[0;32mI (759) psram: This chip is ESP32-D0WD[0m

YanMinge, 2020-09-07:
not esp_audio_play function, call esp_audio_stop, crash appears once in dozens of times, I also got this issue

/home/myan/project/mata_esp32/submodule/esp-idf/components/freertos/queue.c:729 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40095200 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x4009b708:0x3f900f90 |<-CORRUPTED

JosephTang, 2020-09-30:
@tongjinlv Could you provide more information listed below:
1. Hash of ADF
2. Hash of IDF
3. Hash of MicroPython
4. Steps to reproduce this issue
5. log(from boot to crash).

Thanks

jason-mao, 2020-12-02:
@YanMinge How about this issue?

closed 2021-09-26 by jason-mao