espressif/esp-adf #459: ESP32-LyraTD-MSC play_mp3_control audio distorted

andriisemenov, 2020-09-07:
This example plays distorted on ESP32-LyraTD-MSC. Also tried to apply the solution in #297 to set CPU frequency to 240MHz and SPI config to 80MHz, however this does not resolve the issue. BTW, is the correct SPI config located here ( menuconfig->Serial flasher config -> Flash SPI speed )? [audio sample](https://drive.google.com/file/d/1xngJsBjSQPCnHdMSSGVQUN-uMijN-vPT/view?usp=drivesdk)

jason-mao, 2020-09-29:
@andriisemenov Sorry, I can't get the audio sample as your  previous  link. Please attach your sdkconfig file and music sample at here.

maizezoidberg, 2020-11-10:
@andriisemenov do you have any news about this issue? I have some problems

maizezoidberg, 2020-11-10:
@jason-mao just play mp3 file from server, with high ping. In standard play http example

andriisemenov, 2020-11-10:
> @andriisemenov do you have any news about this issue? I have some problems

Not really, I could not fix it. play_mp3 works well for me, but as soon as I load play_mp3_control the audio is distorted. This also happens if I change the link to my own local server (low ping). I think the issue is related to the logic of checking the button states in the loop of main. I've experimented with this library, but shifted to another approach. I will try to attach the sdkconfig here later.