espressif/esp-adf #461: Issue：DLNA Demo在播放一段时间音频流后出现卡顿，不久后重启

Skylakc, 2020-09-13:
Environment：
ESP-IDF 4.1
ESP-ADF 2.1
Ubuntu 20.04
Python 3.8.2

关键log如下
[image]

在正常启动DLNA音频流后一段时间
当出现
[image]
若不刷新重启DLNA流，则会引发崩溃，继而自动重启

请问如何解决

jason-mao, 2020-09-29:
@Skylakc 这个先分析下HTTP为啥得到了500 的status. Crash的话加些打印，跟踪具体挂在什么位置。

kylongmu, 2020-10-01:
我的DLNA测试，按官方步骤来，但是Universal Media Center只能看到设备，点播放键后没反应，然后设备变灰色。换网易云音乐能看到设备，并且耳机听到声音，结果是可怕的卡顿，一秒钟一下那种，感觉是MCU上的接收缓存机制有问题，耳机能听到的声音底噪也高Codec_3V3，看到用了LP2985A-33DBVT来做模拟电源应该没问题，不过我考虑另外开个issue描述下。

Skylakc, 2020-10-08:
> 我的DLNA测试，按官方步骤来，但是Universal Media Center只能看到设备，点播放键后没反应，然后设备变灰色。换网易云音乐能看到设备，并且耳机听到声音，结果是可怕的卡顿，一秒钟一下那种，感觉是MCU上的接收缓存机制有问题，耳机能听到的声音底噪也高Codec_3V3，看到用了LP2985A-33DBVT来做模拟电源应该没问题，不过我考虑另外开个issue描述下。

关于你说的底噪问题我倒是没有遇到过。推荐用Foobar2000+DLNA插件测试，缓冲区可以调大些。不过我在尝试播放96KHz采样率MP3的时候估计遇到处理器性能瓶颈，产生较大卡顿。

Skylakc, 2020-10-08:
> @Skylakc 这个先分析下HTTP为啥得到了500 的status. Crash的话加些打印，跟踪具体挂在什么位置。

谢谢，明天我回学校了继续调试一下

Skylakc, 2020-11-24:
[image]
[image]
[image]
[image]
更新：近期我发现espadf和idf都发生了更新，在更新以后 http500的问题没有再发生了，但是发生了Audio_MEM问题

closed 2022-02-08 by jason-mao