espressif/esp-adf #463: Audio codec input source

utkutpcgl, 2020-09-22:
How can we change the active microphone input to be auxiliary input on esp lyrat? I see that it is on adc line 2, however, I could not find any directives on esp adf documentation.

Thanks, for your awesome library and for your help.

Gaochenchang, 2020-09-29:
If you want to use AUX-IN, you need to modify the es8388 driver. Specifically speaking is to change the value of ES8388_ADCCONTROL2 register in es8388.c into 0x54 and the value of ES8388_ADCCONTROL3 register is 0x83.

utkutpcgl, 2020-09-30:
Thanks! I have solved this problem by changing audio_hal_codec_config_t (.adc_input =AUDIO_HAL_ADC_INPUT_LINE1), but now I face an issue with encoding speed. 
I want to use amr wb encoder rather than nb encoder. Therefore, I increase the i2s sample rate form 8k to 16k. However, the encoding speed was too low and the encoder could not catch up with the i2s_stream_read(16k). How can I increase the audio codec encoding decoding speed or amr encoding speed? Might I be missing something else?

utkutpcgl, 2020-10-17:
Thanks, I will show the result when I'm done.

jujax, 2020-10-22:
Hello,
I use my ESP32-LyraT to send line input audio to bluetooth headset. I opened an issue to explain that : #485

utkutpcgl, 2020-11-25:
Okay, it works. Thanks.

closed 2020-11-25 by utkutpcgl