espressif/esp-adf #464: Microphone listens to Speaker problem (Need sound filter)

utkutpcgl, 2020-09-23:
I use i2s-stream to read the sound detected by the onboard microphone (esp32 Lyrat board) and play the sound with i2s write with an external speaker connected to the onboard speaker output. Also, I only use one channel to listen and play (channel left).

I have the following pipeline: 
**microphone -> i2s_stream_read -> amr_encoder ->amr_decoder ->i2s_stream_write -> speaker** 
(There is approx. 150 ms delay btw. listening and playing the sound due to read-write delays inside the pipeline.)

The microphone listens to the speaker, hence, it enters a loop and repeats to play the previous sound that it has listened to. It can even produce a high-intensity noise due to amplification when the output sound volume is set too high. Otherwise, it repeats the sound it listened once, but it is like an undesired echo that I would like to avoid.

I would like to learn how to overcome this problem. It would be good to use a software filter to filter out the speaker's sound, but I could not find the library for this purpose.  

Thanks for your help!

jason-mao, 2020-09-30:
@utkutpcgl As your description, I think the `algo_stream_init` could be improve your performance.

utkutpcgl, 2020-09-30:
Thanks! Can I use it with ESP32-LyraT?

jason-mao, 2020-10-09:
@utkutpcgl Yes. The algorithm stream ALGORITHM_STREAM_INPUT_TYPE2 can be applied.

utkutpcgl, 2020-10-11:
Thanks again!

frj2a, 2020-10-13:
The "algorithm_stream" that uses "esp_aec" is most probably the way to go.
But could not find how to set it up properly.  It needs to be inserted in the "input" pipeline but also needs a reference ring-buffer from the "output" pipeline. This last part is the one I could not find out how to set it up.

Patsjemoe, 2020-10-14:
Hi,
Thank you for your feedback.
Found esp_aec.h in components\audiostream\algorithm_stream but cannot find
it in the Voip example, despite espressif confirmed it is in it (see
attached pictures).
Any other example using it?
Actually I am not sure how to implement it, so any good suggestion  is
welcome.
Regards
Ludo

jason-mao, 2020-10-15:
Similar issue link #476 

We will provide a correlative example for demonstrate how to use algorithm_stream.

Patsjemoe, 2020-10-15:
Hi,
That would be great.
Thanks
Regards
Ludo

Patsjemoe, 2020-10-21:
Hi Jason Mao,
Any idea when this would come? Espressif documents show it already exists, so I am wondering why I cannot find any example
Thank you in advance
Regards

utkutpcgl, 2020-10-23:
Hello and thanks again @jason-mao , I guess I faced a bug.
When I tried to build the project with algo stream added I face the following error:

`[5/7] Generating ld/esp32.project.ld
warning: the default selection ESP32_KORVO_DU1906_CODEC_TAS5805M (undefined) of <choice ESP32_KORVO_DU1906_DAC> (defined at C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/audio_board/Kconfig.projbuild:27) is not contained in the choice
[6/7] Linking CXX executable dinleme_deneme.elf
FAILED: dinleme_deneme.elf
cmd.exe /C "cd . && C:\Users\utkum\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\dinleme_deneme.elf.rsp  -o dinleme_deneme.elf  && cd ."
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x48): undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x4c): undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x58): undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_noisemean_cal':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:123: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:131: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_pro':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:217: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:257: undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `esp_ns_alloc':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.h:30: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_init':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:295: undefined reference to `esp_kiss_fftr_alloc'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

As far as I understand libesp_audio_processor.a has a reference to **esp_kiss_fftr** in _noise suppression related functions_ in the **ns_core** file.

I use:

> Latest adf version
> idf version 4.1
> Windows idf command prompt
> esp32_lyrat board.

Have a nice day.

Patsjemoe, 2020-10-23:
> The "algorithm_stream" that uses "esp_aec" is most probably the way to go.
> But could not find how to set it up properly. It needs to be inserted in the "input" pipeline but also needs a reference [...]

Hi, 
Could you share already how to define the input part?
Thanks

utkutpcgl, 2020-10-24:
Hello @Patsjemoe ,
I am not certainly sure but in this pipeline:
**microphone -> i2s_stream_read -> (input_buffer1) ->  amr_encoder ->amr_decoder ->(input_buffer2) ->i2s_stream_write -> speaker**
I expect the record signal input buffer to be input_buffer1 and the reference signal's buffer to be input_buffer2. If I am wrong please correct me @jason-mao . This seemed logical to me by looking at the following diagram: 
[image]

Becuase, as far as I know AEC needs an output signal to cancel out the echo, just like @frj2a  said.

jason-mao, 2020-10-27:
> Hi Jason Mao,
> Any idea when this would come? Espressif documents show it already exists, so I am wondering why I cannot find any example
> Thank you in advance
> Regards

@Patsjemoe It will plan to valid on next week.

utkutpcgl, 2020-10-28:
Just in case you missed my question above @jason-mao [https://github.com/espressif/esp-adf/issues/464#issuecomment-715164315](url). Thanks.

Patsjemoe, 2020-11-08:
Hi, did not find any example code yet

Patsjemoe, 2020-11-13:
> > Hi Jason Mao,
> > Any idea when this would come? Espressif documents show it already exists, so I am wondering why I cannot find any example
> > Thank you in advance
> > Regards
> 
> @Patsjemoe It will plan to valid on next week.

@jason-mao : any progress/idea when it will come?
I am trying to find out myself, but the documentation is not sufficiently complete, so more questions then solution

jason-mao, 2020-11-16:
@Patsjemoe Yes, we are working on this. There are something need to  modify for process normally case.

Patsjemoe, 2020-11-16:
OK thank you.

jason-mao, 2020-11-19:
@Patsjemoe @frj2a  ebb87841016bf0b857a72af119b0d33f6039202b is the algorithm stream example.

closed 2020-12-02 by jason-mao