espressif/esp-adf #465: Connecting the ssd1306 monitor to the ESP32-LyraT V4.3 board.

rn6ljk, 2020-09-25:
Hello to all colleagues!
ESP32-LyraT V4.3 board.
Example PLAY_MP3.
The possibility of connecting the ssd1306 monitor to the i2c bus on the LyraT V4.3 board is being studied.
At the beginning of the playback of a fragment of mp3, the text "S O U N D    S T A R T" should be displayed on the monitor.
After playing mp3, the text “F I N I S H    S O U N D” should be displayed on the monitor.
 SSD1306 is attached to I2C Header / JP5.
The i2c driver and ssd12306 display are initializing.
at the beginning of app_main after which the display shows the text - "S O U N D  S T A R T".

…………..
#define SDA_PIN GPIO_NUM_18
#define SCL_PIN GPIO_NUM_23
…………..
void app_main(void)
{
	char tout[32]="     S O U N D\n     S T A R T\n";
	char tout2[32]="\n\n F I N I S H\n  S O U N D\n";
	i2c_master_init();
	ssd1306_init();
…………………………………………………………

	ssd1306_display_pattern();
	ssd1306_clear();
	ssd1306_display_text(tout);
……………………………………
	i2c_config_t i2c_config = {
		.mode = I2C_MODE_MASTER,
		.sda_io_num = SDA_PIN,
		.scl_io_num = SCL_PIN,
		.sda_pullup_en = GPIO_PULLUP_ENABLE,
		.scl_pullup_en = GPIO_PULLUP_ENABLE,
		.master.clk_speed = 1000000
	};
	i2c_param_config(I2C_NUM_0, &i2c_config);
	espRc=i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0);
	if (espRc == ESP_OK) {
			ESP_LOGI(tag_i2c, "I2C  configured successfully");
		} else {
			ESP_LOGE(tag_i2c, "I2C configuration failed. code: 0x%.2X", espRc);
……………………………………………
i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (0x3C << 1) | I2C_MASTER_WRITE, true);
	i2c_master_write_byte(cmd, 0x00, true);
………………………….
………………………..
…………………………
	i2c_master_write_byte(cmd, 0xAF, true);
	i2c_master_stop(cmd);
	espRc = i2c_master_cmd_begin(I2C_NUM_0, cmd, 10/portTICK_PERIOD_MS);
		ESP_LOGI(tag_i2c, "OLED configured successfully");
	} else {
		ESP_LOGE(tag_i2c, "OLED configuration failed. code: 0x%.2X", espRc);
	}
	i2c_cmd_link_delete(cmd);
………………………………………………………….
Installing the i2c driver and configuring the ssd1306 is done without errors:
I (23) I2C install: I2C  configured successfully
I (23) I2C install: OLED configured successfully
The text "S O U N D S T A R T" is successfully displayed on ssd1306.
After executing the snippet:
audio_pipeline_handle_t pipeline;
    audio_element_handle_t i2s_stream_writer, mp3_decoder;
    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_INFO);
the message is displayed:
I (93) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip
E (93) i2c: i2c driver install error
E (93) I2C_BUS: C:/Users/RN6LJK/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
E (103) I2C_BUS: C:/Users/RN6LJK/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
E (113) I2C_BUS: C:/Users/RN6LJK/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
……………………………………………..
……………………………………………..
E (393) I2C_BUS: C:/Users/RN6LJK/esp-adf/components/esp_peripherals/driver/i2c_bus/i2c_bus.c:86 (i2c_bus_write_bytes):Handle error
E (403) AUDIO_HAL: codec init failed!
E (403) AUDIO_BOARD: C:/Users/RN6LJK/esp-adf/components/audio_board/lyrat_v4_3/board.c:55 (audio_board_codec_init): Got NULL Pointer
E (423) AUDIO_HAL: audio_hal handle is null
I (423) PLAY_MP3_FLASH: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event
I (433) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
I (443) PLAY_MP3_FLASH: [2.2] Create i2s stream to write data to codec chip
I (453) PLAY_MP3_FLASH: [2.3] Register all elements to audio pipeline
I (463) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (473) PLAY_MP3_FLASH: [ 3 ] Set up  event listener
I (473) PLAY_MP3_FLASH: [3.1] Listening event from all elements of pipeline
I (483) PLAY_MP3_FLASH: [ 4 ] Start audio_pipeline
I (503) PLAY_MP3_FLASH: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
I (7253) PLAY_MP3_FLASH: [ 5 ] Stop audio_pipeline
The piece of music is played, after which a message is displayed:
W (7253) AUDIO_ELEMENT: [mp3] Element already stopped
W (7253) AUDIO_ELEMENT: [i2s] Element already stopped
W (7263) AUDIO_PIPELINE: There are no listener registered
W (7263) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (7273) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
W (7253) AUDIO_ELEMENT: [mp3] Element already stopped
W (7253) AUDIO_ELEMENT: [i2s] Element already stopped
W (7263) AUDIO_PIPELINE: There are no listener registered
W (7263) AUDIO_ELEMENT: [i2s] Element has not create when AUDIO_ELEMENT_TERMINATE
W (7273) AUDIO_ELEMENT: [mp3] Element has not create when AUDIO_ELEMENT_TERMINATE
The example ends with the code:
espRc = i2c_master_cmd_begin(I2C_NUM_0, cmd, 10/portTICK_PERIOD_MS);
		if (espRc == ESP_OK) {
			ESP_LOGI(tag_i2c, "OLED pattern successfully");
		} else {
			ESP_LOGE(tag_i2c, "OLED pattern failed. code: 0x%.2X", espRc);
		}
		i2c_cmd_link_delete(cmd);
Then a message is displayed:
E (8293) I2C install: OLED pattern failed. code: 0x107
The text "F I N I S H S O U N D" is successfully displayed on ssd1306.
Similar errors appear when connecting sd1306 to LyraT V4.3 board and executing VOIP, PLAY_MP3, wifi_bt_ble_coex examples.
If you specify the I2C_NUM_1 parameter when installing the i2c driver, then the specified error does not appear. In my opinion, this is not a solution to the problem.
In the VOPI example, when the monitor ssd13406 is connected to the i2c bus, an error appears:
C:/Users/RN6LJK/esp-idf/components/freertos/queue.c:1443 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40089925 on core 0

ELF file SHA256: 895b63434a320f65e686dc735c53e36adcf5e7c5a53085091db30b79765a9abc

Backtrace: 0x4008d70c:0x3ffccdc0 0x4008d955:0x3ffccde0 0x40089925:0x3ffcce00 0x40082481:0x3ffcce40 0x400d4e49:0x3ffcce70 0x400d4801:0x3ffccec0 0x400d49a2:0x3ffccee0 0x40176373:0x3ffccf10 0x400e5246:0x3ffccf30 0x400e5606:0x3ffccf60 0x400e595e:0x3ffccf80 0x400e614e:0x3ffccfa0
Who faced this problem?
Please share your results.
Thanks.

jason-mao, 2020-09-29:
@rn6ljk Could you feedback the backtrace as follow command `xtensa-esp32-elf-addr2line -pfia -e YOUR_ELF_FILE 0x4008d70c:0x3ffccdc0 0x4008d955:0x3ffccde0 0x40089925:0x3ffcce00 0x40082481:0x3ffcce40 0x400d4e49:0x3ffcce70 0x400d4801:0x3ffccec0 0x400d49a2:0x3ffccee0 0x40176373:0x3ffccf10 0x400e5246:0x3ffccf30 0x400e5606:0x3ffccf60 0x400e595e:0x3ffccf80 0x400e614e:0x3ffccfa0`?

rn6ljk, 2020-10-12:
Hi colleague.
Did I fulfill your request correctly?

C:\Users\RN6LJK\eclipse-workspace\play_mp3_1306_i2c\build>xtensa-esp32-elf-addr2line -pfia -e play_mp3.elf 0x4008d70c:0x3ffccdc0 0x4008d955:0x3ffccde0 0x40089925:0x3ffcce00 0x40082481:0x3ffcce40 0x400d4e49:0x3ffcce70 0x400d4801:0x3ffccec0 0x400d49a2:0x3ffccee0 0x40176373:0x3ffccf10 0x400e5246:0x3ffccf30 0x400e5606:0x3ffccf60 0x400e595e:0x3ffccf80 0x400e614e:0x3ffccfa0
0x4008d70c: ?? ??:0
0x4008d955: ?? ??:0
0x40089925: prvReceiveGenericFromISR at C:/Users/RN6LJK/esp-idf/components/esp_ringbuf/ringbuf.c:969
0x40082481: heap_caps_realloc at C:/Users/RN6LJK/esp-idf/components/heap/heap_caps.c:232
0x400d4e49: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1199 (discriminator 118)
0x400d4801: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1018 (discriminator 4)
0x400d49a2: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1063 (discriminator 12)
0x40176373: ?? ??:0
0x400e5246: pvmp3_get_side_info at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-mp3/src/pvmp3_get_side_info.cpp:163
0x400e5606: pvmp3_huffman_parsing at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-mp3/src/pvmp3_huffman_parsing.cpp:255 (discriminator 1)
0x400e595e: pvmp3_dequantize_sample at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-mp3/src/pvmp3_dequantize_sample.cpp:243
0x400e614e: pvmp3_polyphase_filter_window at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/codec-utils/include/common_macros.h:252 (discriminator 2)
 (inlined by) fxp_mac32_Q32 at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-mp3/src/pv_mp3dec_fxd_op_esp32.h:79 (discriminator 2)
 (inlined by) pvmp3_polyphase_filter_window at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-mp3/src/pvmp3_polyphase_filter_window.cpp:215 (discriminator 2)

rn6ljk, 2020-10-12:
Traceback (most recent call last):
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 805, in <module>
    main()
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 717, in main
    monitor.main_loop()
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 369, in main_loop
    (event_tag, data) = self.event_queue.get()
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\queue.py", line 170, in get
    self.not_empty.wait()
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 296, in wait
    waiter.acquire()
KeyboardInterrupt
Traceback (most recent call last):
  File "C:\Users\RN6LJK\esp-idf\tools\idf.py", line 642, in <module>
    main()
  File "C:\Users\RN6LJK\esp-idf\tools\idf.py", line 621, in main
    execute_action(actions[0], actions[1:])
  File "C:\Users\RN6LJK\esp-idf\tools\idf.py", line 615, in execute_action
    function(action, args)
  File "C:\Users\RN6LJK\esp-idf\tools\idf.py", line 343, in monitor
    _run_tool("idf_monitor", monitor_args, args.project_dir)
  File "C:\Users\RN6LJK\esp-idf\tools\idf.py", line 86, in _run_tool
    subprocess.check_call(args, env=os.environ, cwd=cwd)
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 325, in call
    return p.wait(timeout=timeout)
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 990, in wait
    return self._wait(timeout=timeout)
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1230, in _wait
    timeout_millis)

rn6ljk, 2020-10-12:
C:\Users\RN6LJK\eclipse-workspace\play_mp3_1306_i2c>Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 134, in _run_outer
    self.run()
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 228, in run
    data = self.serial.read(self.serial.in_waiting or 1)
  File "C:\Users\RN6LJK\.espressif\python_env\idf3.3_py3.7_env\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Unhandled exception in thread started by <bound method Thread._bootstrap of <Thread(Thread-2, started 12056)>>
Traceback (most recent call last):
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 917, in _bootstrap_inner
    self.run()
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 134, in _run_outer
    self.run()
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 228, in run
    data = self.serial.read(self.serial.in_waiting or 1)
  File "C:\Users\RN6LJK\.espressif\python_env\idf3.3_py3.7_env\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, '

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 885, in _bootstrap
    self._bootstrap_inner()
  File "C:\Users\RN6LJK\AppData\Local\Programs\Python\Python37\lib\threading.py", line 927, in _bootstrap_inner
    (self.name, _format_exc()), file=_sys.stderr)
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 798, in write
    self._output_write(b)
  File "C:\Users\RN6LJK\esp-idf\tools/idf_monitor.py", line 759, in _output_write
    self.output.write(data.decode())
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 0: unexpected end of data

jason-mao, 2020-12-02:
@rn6ljk How about this problem?

closed 2022-02-08 by jason-mao