espressif/esp-adf #466: Compile example "pipeline_a2dp_source_stream", get 4 error messages, without any code by myself!

kylongmu, 2020-09-27:
ESP-IDF 4.3
ESP-ADF 2.1
--------------------------------------Here is the error message!----------------------------

1. CMake Error at C:/Users/THINKPAD/.espressif/esp-idf/tools/cmake/build.cmake:271 (message):Some Python dependencies must be installed.  Check above message for
details.
-play_bt_source_example.c
2. 'memcpy' forming offset [22, 249] is out of the bounds [0, 21] [-Werror=array-bounds]
3.too few arguments to function 'esp_bt_gap_set_scan_mode'
4.'ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE' undeclared (first use in this function); did you mean 'ESP_BT_GENERAL_DISCOVERABLE'?

kylongmu, 2020-09-27:
Oh my GOD!
The ADF only support IDF 3.3 version.

kylongmu, 2020-09-28:
After change the IDF to version 4.1, I can build succesfully, but it's hard.
The 4 errors are still, comment the first 3 error, then I found some info:
"Some Python dependencies must be installed.  Check above message for details."
After a lot of test, I found it's caused by an python option: in this file
idf_tools.py
`line 1185            subprocess.check_call([sys.executable, '-m', 'pip', 'install', '--user', 'virtualenv']`

Remove --user option
Then redo install and export under IDF folder
build success!

kylongmu, 2020-09-28:
So all the 4 errors, 3 comes from example code, 1 come from IDF install problem.
Please fix these bugs!

jason-mao, 2020-09-29:
@kylongmu Thanks for you report, the fix is coming.  Please note that ADF can be support IDF released version only,  ESP-IDF 4.3 have not released.

kylongmu, 2020-09-29:
@jason-mao Thanks for your response, it's nice for the problem to be notified.

closed 2020-09-30 by espressif-bot