espressif/esp-adf #469: Compile example "dlna", got a link error:undefined reference to '_binary_devicelist_xml_start'. without and code by myself.

kylongmu, 2020-09-29:
ESP-IDF 4.1
ESP-ADF 2.1
-----------------------------------
```
[2/6] Performing build step for 'bootloader'
ninja: no work to do.
[3/4] Linking CXX executable dlna_example.elf
FAILED: dlna_example.elf
cmd.exe /C "cd . && C:\Users\THINKPAD\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address -g   @CMakeFiles\dlna_example.elf.rsp  -o dlna_example.elf  && cd ."
c:/users/thinkpad/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: 
D:/Develop/Git_Project/esp-adf/components/esp-adf-libs/esp_upnp/lib/esp32\libesp_upnp.a(upnp.o):(.literal.upnp_serve_root_files+0x1c): undefined reference to `_binary_devicelist_xml_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.
```

It looks like the '_binary_devicelist_xml_start' losted, I tried to do this under ADF path:
` git submodule update --init --recursive`
But the problem still, so I need help again!
Why so many problems in the examples, it's stopping people who want to try this chip.

jason-mao, 2020-09-30:
@kylongmu Thank you for your report. Let's checking.

kylongmu, 2020-09-30:
@jason-mao Thanks for your response, I think maybe all the examples should be checked.

closed 2020-09-30 by espressif-bot