espressif/esp-adf #470: I2S + AMR-WB does not work with 16K sample rate.

utkutpcgl, 2020-09-30:
This is my pipeline:
- [codec_chip] -> i2s_stream_reader -> ringbuf1 -> amr_encoder -> ringbuf2 -> amr_decoder -> i2s_stream_writer -->[codec_chip]

Although AMR-NB works fine with i2s-sample-rate is 8000, AMR-WB does not work in the pipeline above when i2s-sample-rate is 16000. These were the configurations given in the available AMR-WB examples in esp-adf. The only difference is that I decode the encoded data directly to play it right away from the speaker.
I start the codec with : 

    audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);
And create an element with the following configurations.
    
    i2s_stream_cfg_t i2s_cfg_reader = I2S_STREAM_CFG_DEFAULT(); // configuration for the i2s_stream_reader
    i2s_cfg_reader.i2s_config.sample_rate = 16000;
    i2s_cfg_reader.i2s_config.channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT; //I2S_CHANNEL_FMT_RIGHT_LEFT
    i2s_cfg_reader.type = AUDIO_STREAM_READER;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg_reader);

I2S reads from the microphone and writes to the speaker directly. The sound is continuous when I use AMR-NB with an 8000 sample rate, however, it is discontinuous when the sample rate is 16000 (laggy). I thought that the codec speed was insufficient, hence, I tried to increase it by making audio_hal_codec_i2s_iface_t.samples = AUDIO_HAL_16K_SAMPLES. However, it had no effect. 

I have an ESP32-LyraT board and I use IDF-v4.1 and latest-version ADF.

jason-mao, 2020-10-13:
@utkutpcgl Please change the sdkconfig file as follow

```c
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_FREERTOS_HZ=1000

CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_ESP32_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_SPEED_80M=y

```

utkutpcgl, 2020-10-27:
Thanks, I will show the results later.

utkutpcgl, 2020-11-25:
After I made those changes it works! Thanks.

closed 2020-11-25 by utkutpcgl