espressif/esp-adf #471: DLNA example, without any code by myself, can't find the dlna device by 'Universal Media Center '. Console output some error messages.

kylongmu, 2020-09-30:
ESP_IDF 4.1
ESP_ADF 2.1
Board: esp32-lyrat-v4.3
------------------------------------Output of console port------------------------------------------
```
[0;33mW (7039) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4[0m
I (7109) wifi:AP's beacon interval = 102400 us, DTIM period = 3
[0;32mI (11049) PERIPH_WIFI: Got ip:192.168.50.105[0m
[0;32mI (11049) esp_netif_handlers: sta ip: 192.168.50.105, mask: 255.255.255.0, gw: 192.168.50.1[0m
[0;32mI (11049) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;31mE (11049) gpio: gpio_install_isr_service(438): GPIO isr service already installed[0m
[0;32mI (11059) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (11059) ES8388_DRIVER: init,out:02, in:00[0m
[0;32mI (11059) AUDIO_HAL: Codec mode is 2, Ctrl:1[0m
[0;32mI (11069) AUDIO_THREAD: The media_task task allocate stack on external memory[0m
[0;31mE (11069) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first
[0m
[0;31mE (11069) AUDIO_THREAD: Error creating RestrictedPinnedToCore media_task[0m
[0;31mE (11069) ESP_AUDIO_CTRL: Error create media_task[0m
[0;32mI (11069) ESP_DECODER: esp_decoder_init, stack size is 5120[0m
[0;32mI (11069) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3[0m
[0;32mI (11069) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3[0m
[0;32mI (11089) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0[0m
[0;32mI (11089) LYRAT_V4_3: I2S0, MCLK output by GPIO0[0m
[0;31mE (11089) ESP_AUDIO_CTRL: /Users/maojianxin/duros/esp-adf-libs-source/esp_audio/src/media_ctrl.c:1338 (media_ctrl_callback_set): Got NULL Pointer[0m
[0;32mI (11089) AUDIO_THREAD: The ssdp_task task allocate stack on external memory[0m
[0;31mE (11089) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v3.3_freertos.patch" first
[0m
[0;31mE (11089) AUDIO_THREAD: Error creating RestrictedPinnedToCore ssdp_task[0m
[0;31mE (11089) SSDP: not enough memory?[0m
[0;33mW (11089) SSDP: SSDP socket is -1, ignore socket_close request.[0m
[0;32mI (11089) SSDP: ssdp_stop[0m
[0;32mI (11099) DLNA_EXAMPLE: DLNA Started...[0m
```
At this moment I can build the example and download it, but the output looks like have some errors, please give more test to make the board work well.

kylongmu, 2020-10-01:
Okay, when I tryed this, I can find the DLNA device:
`git apply D:\Develop\Git_Project\esp-adf\idf_patches\idf_v3.3_freertos.patch`
Why hide the key message in the device console output, is't a crazy joke?

closed 2020-10-01 by kylongmu

xiaodeme-aha, 2023-09-18:
Mine is the same. Do you have a suitable solution now

xiaodeme-aha, 2023-09-18:
I know that this statement indicates that the ssdp service is not fully turned on,�[0;33mW (11089) SSDP: SSDP socket is -1, ignore socket_close request.�[0m