espressif/esp-adf #472: DLNA测试：耳机输出声音小底噪高；音流卡顿；Universal Media Center只能看到设备，点播放键后没反应，然后设备变灰色，网易云音乐能播放。

kylongmu, 2020-10-01:
3个问题，我的DLNA测试，按官方步骤来：
（1）Universal Media Center只能看到设备，点播放键后没反应，然后设备变灰色。换网易云音乐能看到设备，并且耳机听到声音。
（2）单音乐流是可怕的卡顿，一秒钟一下那种，感觉是MCU上的接收缓存机制有问题。
（3）耳机能听到的声音很小，底噪也高，检查了Codec_3V3，看到用了LP2985A-33DBVT来做模拟电源应该没问题，我的耳机可是非常好的，怀疑开发板耳机电路接的有毛病。
[image]
四个脚名找ES8388输出：
[image]
以上两个图片来自官方电路图耳机输出，ES8388输出脚。
感觉这耳机输出接法奇怪：
（1）电容串联2次22uF到耳机，手册里建议是串联1个47uF，电容串联好像更小了吧？
（2）耳机驱动没有按LOUT1, LOUT2一组，而是把LOUT1的前级DAC_LOUT1与自己接，有这种接法的吗？

kylongmu, 2020-10-01:
换4W小音箱来测试，听不到底噪：
[image]
从图上看，音箱输出LOUT1, LOUT2一组，ROUT1, ROUT2一组，各驱动一个音箱，怀疑你们的耳机输出真接错了，天啊，软件bug一堆就算了，现在看还有硬件bug！

kylongmu, 2020-10-01:
继续测试音流卡顿问题，用手机wifi建立热点，ESP32直连手机热点，DLNA播放流畅。换到开发板、手机都连无线AP，音流卡顿又出现，这个看来真是缓存问题了，中间只是多了个无线AP。同样经过无线AP测试用手机连自己电脑的DLNA设备没这个问题，说明无线AP没有影响电脑的DLNA流。

kylongmu, 2020-10-12:
看到乐鑫新推的ESP-LyraT-8311A扩展板上面的耳机输出电路，果然与esp32-lyrat-4.3上的8388不一样：
[image]
[image]
OUTP\OUTN一对输出直推，再看8388的耳机推动电路，应该是错了，我换了耳机试验esp32-lyrat-4.3的还是输出底噪高、音量小。

ESP-NJ, 2020-10-13:
Kylongmu：
您好！
        关于您提到的ES8388的耳机设计疑问，答复如下：
1） 关于串联了两个22uF电容：此处硬件是兼容设计，贴一个电容即可，开发板上默认两个电容都贴片了。电容容值影响滤波器的低频截止频率，对低频性能有一定影响，当前的截止频率为220Hz左右，如果您需要更低的截止频率，可以增大电容的容值，或者去掉一个电容，直接短接。
2） 关于ES8388耳机硬件设计，您可能有些误解，关于LOUT1和DAC_LOUT1，其中DAC_LOUT1是用于后级喇叭PA做输入的，当耳机未插入时，LOUT1和DAC_LOUT1是短接的，此时声音从喇叭输出，当耳机插入后LOUT1和DAC_LOUT1是断开的，声音从耳机输出，喇叭没有声音。而ES8388的LOUT1，LOUT2是两路独立的信号，并不是差分信号，详细可以看ES8388的规格书，在做喇叭PA输入时，两路信号一起做了伪差分设计，而耳机部分受限于耳机接口管脚限制，只是单端设计。
3）您提到的耳机声音小问题，可以尝试按音量键调节，开发板耳机（32欧姆负载）最大输出功率为20mW左右，只适用于一些耳塞式入耳式等小功率耳机，不确定是否与您的耳机匹配。关于底噪高问题，我这边并没有复现，需要您再提供详细的测试场景，及耳机型号信息等。

谢谢！
NJ
Espressif

kylongmu, 2020-10-13:
> Kylongmu：
> 您好！
> 关于您提到的ES8388的耳机设计疑问，答复如下：
> 1） 关于串联了两个22uF电容：此处硬件是兼容设计，贴一个电容即可，开发板上默认两个电容都贴片了。电容容值影响滤波器的低频截止频率，对低频性能有一定影响，当前的截止频率为220Hz左右，如果您需要更低的截止频率，可以增大电容的容值，或者去掉一个电容，直接短接。
> 2） [...]

感谢您的回复，截至频率220Hz太高了，应该到1kHz的频率增益曲线都不好看吧，只有自己动手换电容了。单端输出，只有20mW推动力，我明白了，我的耳机都是比较好的耳机需要推动力强点的，这个输出推动实在太小，我调到最大音量就感觉底噪很高声音还是小。

closed 2020-10-15 by jason-mao