espressif/esp-adf #473: ESP32 Lyrat right speaker has less volume than left speaker

utkutpcgl, 2020-10-04:
I tried to make the volume maximum in the audio HAL configurations and ALC. Also, I changed most of the register values in ES8388 to obtain an equal sound level on the left and right speaker output and microphone input. I could link the right microphone input to the right speaker and left microphone to the left speaker, however, I was unsuccessful in increasing the volume of the right speaker. I changed the following ES8388 -register values to set the right volume output to be maximum: 4,9,11,12,18,26,27,29,38,39,42,47. Now, all of the left and right configurations should be the same, but nothing changed. I suspect if there is a hardware issue or sth. I am missing.

Thanks in advance.

Gaochenchang, 2020-10-22:
According to the es8388 datasheet, the adjustable gain range is 0 dB ~ 24 dB, and the gain interval is 3. For this API, the actual available range is also 0~24. The value interval is 3.

utkutpcgl, 2020-11-28:
When I set the gain with es8388_set_mic_gain(24) the gain is less than the default gain. 

[image]
[https://github.com/espressif/esp-adf/issues/177](url)

Is this true?

I want to set the right-left microphone gain and speaker level to the maximum with equal output. 
Setting 
`es8388_set_mic_gain(24)`  
and 
`audio_hal_set_volume(board_handle->audio_hal,100)` 
did not solve my problem. What should  I do?

This means that left-right maximum gain levels are also not equal.

Thanks in advance.

closed 2020-12-02 by espressif-bot

Gaochenchang, 2020-12-02:
You can use the latest master to solve this problem.
Best wishes

utkutpcgl, 2020-12-04:
Thanks @Gaochenchang . Now I can reach the maximum output sound volume with `es8388_set_mic_gain(33)`, 24 still is slightly less than the default gain. So, I guess unlike stated in this comment [https://github.com/espressif/esp-adf/issues/473#issuecomment-714337263](url) , the interval is between 0 to 33. Please correct me if I am wrong.

Gaochenchang, 2020-12-07:
Hello, the maximum mic gain of es8388 is 24dB

utkutpcgl, 2020-12-08:
Dear @Gaochenchang ,

I already know that the datasheet says that the gain is 24dB. However, I have also tested the maximum mic gain with: 
`ESP_ERROR_CHECK(es8388_write_reg(ES8388_ADCCONTROL1, 0b10111011));` // 33 db gain (3*11)
The gain is slightly higher than:
`ESP_ERROR_CHECK(es8388_write_reg(ES8388_ADCCONTROL1, 0b10001000));` // 24 db gain (3*8)

Beforehand, I approved this also with :
`es8388_set_mic_gain(33)` 
is slightly higher than:
`es8388_set_mic_gain(24)` 

I suspect the es8388 datasheet might be missing this information. Still, I might be doing something wrong, however, I would appreciate it if you test this and inform me if I have problems with my codec (es8388) on LyraT. 

Thanks in advance.

Gaochenchang, 2020-12-14:
Hi, es8388 driver is no problem.I set the gain to 33, almost no difference.

utkutpcgl, 2020-12-14:
Hi, thanks. It is almost the same, yes, and there is only a tiny difference.

expresspotato, 2021-04-07:
Look at the commit, this needs to be re-opened.