espressif/esp-adf #474: Using esp_audio_seek in short succession causes xQueueGenericSend assert failed!

dereisl, 2020-10-08:
When using the esp_audio_seek API to seek with a short time between seeks, the ESP will crash with the following error:

```
I (41757) ESP_AUDIO_CTRL: Seek done, new pos:777004
I (42487) ESP_AUDIO_CTRL: Total:2335413, bps:111000, duration:167232 ms
I (42487) ESP_AUDIO_CTRL: Seek with RUNNING status, total bytes:2335413
/Users/dereisl/esp/esp-idf/components/freertos/queue.c:729 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x40097fbf on core 1
0x40097fbf: xQueueGenericSend at /Users/dereisl/esp/esp-idf/components/freertos/queue.c:729 (discriminator 5)
```

Full log: https://gist.github.com/dereisl/1bb3086ad53221eb292c7f936efa1907

I'm using ESP ADF release v2.1 with ESP IDF release v4.0.1 and my audio playing code is similar to https://github.com/espressif/esp-adf/blob/bafffc5ae08c9ed79920b5398cb10cdaa34bfe4a/examples/advanced_examples/dlna/main/dlna_example.c

jason-mao, 2020-10-09:
@dereisl Thanks for your report. I think this issue has been fix on latest master bafffc5. Would you please test it?

dereisl, 2020-10-14:
ESP says:
```
I (4351) DLNA_EXAMPLE: DLNA Started...
I (36961) UPNP_NOTIFY: New Subscribe successful, sid = uuid:5851f42d-8ccf-f646-3129-30705b045db4, timeout = 1800
```

Universal Media Server says:
```
5:56:25 INFO  Universal Media Server is now available for renderers to find
15:57:07 WARN  Couldn't find UPnP service AVTransport for device ESP32 MD (ESP32 Renderer) when trying perform action SetAVTransportURI
15:57:07 WARN  Couldn't find UPnP service AVTransport for device ESP32 MD (ESP32 Renderer) when trying perform action Play
```

Could this be an issue with my configuration aswell? (everything is set to default in UMS though)

dereisl, 2020-10-14:
Seems to be an issue with my UMS config, it plays fine with BubbleUpnp on my Android phone, however, the issue I reported still persists, when I seek very quickly after seeking, the esp crashes with the following log: https://gist.github.com/dereisl/26016e9993fdd6f1a837c42b931087a7

jason-mao, 2020-10-15:
@dereisl 

> however, the issue I reported still persists, when I seek very quickly after seeking, the esp crashes with the following log: https://gist.github.com/dereisl/26016e9993fdd6f1a837c42b931087a7

I have saw the esp_audio version is not the latest master version, just  is  ` ESP_AUDIO-v1.6.3-4-g113c153` on your log , would you please help to test with latest master?

dereisl, 2020-10-15:
I can't get it to compile on the latest master: https://gist.github.com/dereisl/21c9e2152ea96f11a67117360059b954

jason-mao, 2020-10-15:
@dereisl Please run ‘git submodule update --init’ at $ADF_PATH.

dereisl, 2020-10-15:
@jason-mao I already tried that

jason-mao, 2020-10-15:
@dereisl Would you help to test with IDF v4.1? Please help to check is it failed on v4.1

dereisl, 2020-10-15:
@jason-mao https://gist.github.com/dereisl/55eb2863e4b342b6e4983071f7ffdb6b

jason-mao, 2020-10-15:
@dereisl There is warning message `  crosstool-ng version esp-2019r2 doesn't match supported version esp-2020r2.` , IDF 4.1 should update the toolchain.

dereisl, 2020-10-15:
@jason-mao I'm sorry for missing that. I updated the toolchain as well, now I get the same compiler error as with IDF 4.0 again: https://gist.github.com/dereisl/d23d08dac7eb94c36ec1a81f48b2048f

jason-mao, 2020-10-16:
@dereisl OK, I found some strange thing on v4.0 with CMake, which is  libraries dependency is random. v4.0 still support make, if you must be use v4.0, please test with make.

dereisl, 2020-10-16:
@jason-mao I do not have to use v4.0, but with cmake (using `idf.py build`), neither v4.0 nor v4.1 work.
With make I can compile both without issue.
I won't be able to test if the runtime crash still occurs right now, but I will get back to you later.

dereisl, 2020-10-17:
I managed to compile the master branch of esp-adf with both IDF 4.0.1 and IDF 4.1 using `make all` instead of `cmake`/`idf.py build`.
However, the problem is still there:
IDF 4.0.1: https://gist.github.com/dereisl/67d1064f95e66981b18766ca0e316eb8
IDF 4.1: https://gist.github.com/dereisl/478ebb8505c3b3fe68dfb7393a1bd3cf

Steps to reproduce:
1. Connect with BubbleUPnP or any other DLNA controller
2. Play song
3. Seek around quickly
4. ESP crashes and reboots with the above error

jason-mao, 2020-10-19:
@dereisl Thank you for your feedback.Let us checking.

jason-mao, 2020-10-19:
@dereisl Which code do you used? I use the`ADF_PATH/examples/advanced_examples/dlna` can't reproduced your problem with Netease music player.

dereisl, 2020-10-19:
@jason-mao I use the same example, the only thing I changed is that I commented out [these lines](https://github.com/espressif/esp-adf/blob/789998a60d90ac3bba99d0dc869c72ff733d3c5a/examples/advanced_examples/dlna/main/dlna_example.c#L208-L214) because I use a different DAC without volume control.

Maybe try using a file with higher bitrate or seek more frequently? I can try to send a video if that helps.

dereisl, 2020-12-06:
Hello?

jason-mao, 2021-02-23:
@dereisl Hi

> Maybe try using a file with higher bitrate or seek more frequently? I can try to send a video if that helps.

How is the video?

closed 2022-02-08 by jason-mao