espressif/esp-adf #476: Voip echo

Patsjemoe, 2020-10-10:
I am using linphone on android with the Voip in esp-adf. Voice is OK on the lyrat or audiokit, but hear everything as echo of what I say. On the esp32 side, there is no echo.
Could it be that microphone of esp is set too sentitive? How can I adapt this setting? 
Anyone a solution for this?

Patsjemoe, 2020-10-11:
Hi,
I was just wondering if this could be a sort of voice delay instead of echo...? How to attack this?

Patsjemoe, 2020-10-12:
when using linphone with microsip on my pc, NO ECHO
So it looks as it is related to the esp32

Patsjemoe, 2020-10-12:
Do I need to use a different codex like speexx?
Can this be used?

frj2a, 2020-10-13:
There's an "algorithm_stream" that uses "esp_aec" (auto echo cancelation) that probably is the way to go.
But could not find how to set it up propperly.  It needs to be inserted in the "input" pipeline but also needs a reference ring-buffer from the "output" pipeline.

Patsjemoe, 2020-10-27:
Hi Jason-Mao,
You mentioned : We will provide a correlative example for demonstrate how to use algorithm_stream... Any idea when this will come? 
Thank you in advance
Ludo

Patsjemoe, 2020-11-06:
@jason-mao : hi, any progress on the algorithm stream example, planned for this week?
Thanks in advance
Ludo

Patsjemoe, 2020-11-16:
@jason-mao : you had planned the algorithm example two weeks ago. Can you please confirm status?

Patsjemoe, 2020-11-19:
Example received, thank you

closed 2020-11-19 by Patsjemoe