espressif/esp-adf #477: example asr doesn't compiles

Saynok, 2020-10-13:
I have tried git submodule update --recurse it doesn't helps
I'm using visual studio code
this is a log:

`[3/4] Linking CXX executable speech_recognition_example.elf
FAILED: speech_recognition_example.elf
cmd.exe /C "cd . && C:\Users\79080\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-addre
ss   @CMakeFiles\speech_recognition_example.elf.rsp  -o speech_recognition_example.elf  && cd ."
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.model_destroy+0x8): undefined reference to `dl_matrixq
_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.model_create+0x1c): undefined reference to `dl_matrixq
_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.add_speech_commands+0x14): undefined reference to `get
_id_name'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x18): undefined reference 
to `dl_convq_lstm_layer'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x1c): undefined reference 
to `dl_relu_convq'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x20): undefined reference 
to `dl_convq_to_matrix2dq'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x24): undefined reference 
to `dl_matrix2d_from_matrixq'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x28): undefined reference 
to `dl_matrix_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x2c): undefined reference 
to `dl_softmax'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o):(.literal.lvcsr_model_run_accelerate+0x30): undefined reference 
to `dl_matrix_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o): in function `model_destroy':
/home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:257: undefined reference to `dl_matrixq_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:258: undefined reference to `dl_matrixq_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:259: undefined reference to `dl_matrixq_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:272: undefined reference to `dl_convq_queue_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:274: undefined reference to `dl_convq_queue_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:277: undefined reference to `dl_convq_queue_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:278: undefined reference to `dl_convq_queue_free'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libmultinet.a(multinet1_quantized_cn.o): in function `model_create':
/home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:339: undefined reference to `dl_convq_queue_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:345: undefined reference to `dl_matrixq_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:347: undefined reference to `dl_matrixq_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:347: undefined reference to `dl_convq_queue_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:350: undefined reference to `dl_matrixq_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:351: undefined reference to `dl_convq_queue_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 /home/wangwangwang/workspace/AI/esp_sr_lib/components/multinet/multinet1_quantized_cn.c:352: undefined reference to `dl_convq_queue_alloc'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
[... cut ...]
 /home/sunxiangyu/esp/esp_sr_lib/components/wakenet/mfcc_runner_straight.c:72: undefined reference to `csf_logfbank_HTK_step'
c:/users/79080/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
 C:/Users/79080/.espressif/esp-adf/components/esp-sr/lib\libwakenet.a(mfcc_runner_straight.o): in function `run':
ninja: build stopped: subcommand failed.
The terminal process "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build ." terminated with exit code: 1.
`

Saynok, 2020-10-14:
Is esp-idf version requires?

jason-mao, 2020-10-14:
Uhm, there is bug on `esp-sr`, the fix is coming.

closed 2020-10-15 by espressif-bot

jalakamkiranadspl, 2021-07-27:
I am still getting this error any ideas for fix?