espressif/esp-adf #478: Question about wake word customisation and russian language

Saynok, 2020-10-15:
Hello! My name is Igor, and I'm using the esp-adf library in my development. While reading [article](https://github.com/espressif/esp-sr/blob/6370e80fe7028b635303312cb0cfbbe1c4325c8e/wake_word_engine/ESP_Wake_Words_Customization.md) on github, I have misunderstand about the Wake Word Customisation process.
I'm intend to use esp32 mcu and esp-adf library to work with peoples in russian, so is it possible to set as "Wake up word" in russian? I know esp32 has chinese and english phonemic inside, but how to add russian and customize wake word by myself?

Saynok, 2020-10-20:
This is still actual, will be very happy to see an answer!

jason-mao, 2020-10-20:
@Saynok Hello Igor, yes in theory.Firstly, you need to provide the `Wake up word` training voice for us as [requirements](https://github.com/espressif/esp-sr/blob/6370e80fe7028b635303312cb0cfbbe1c4325c8e/wake_word_engine/ESP_Wake_Words_Customization.md#requirements-on-corpus-texts) , then we build the `Wake up word` module library for you. If you have more question about customize wake word please email us by sales@espressif.com.

Saynok, 2020-10-20:
@jason-mao thanks for the answer! Will write to this sales@espressif.com.

closed 2020-10-20 by Saynok