espressif/esp-adf #479: wakeNet5, Core 1 panic'ed

bulazhang, 2020-10-16:
ADF v2.1, IDF 3.3.2
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0

I got the following exception:

`I (41) boot: compile time 13:09:00
I (50) boot: Enabling RNG early entropy source...
I (50) boot: SPI Speed      : 40MHz
I (50) boot: SPI Mode       : DIO
I (51) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (52) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (53) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (54) boot:  2 factory          factory app      00 00 00010000 00300000
I (55) boot: End of partition table
I (56) boot_comm: chip revision: 1, min. application chip revision: 0
I (57) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6cf90 (446352) map
I (209) esp_image: segment 1: paddr=0x0007cfb8 vaddr=0x3ffb0000 size=0x03058 ( 12376) load
I (214) esp_image: segment 2: paddr=0x00080018 vaddr=0x400d0018 size=0x108d80 (1084800) map
I (581) esp_image: segment 3: paddr=0x00188da0 vaddr=0x3ffb3058 size=0x00f94 (  3988) load
I (582) esp_image: segment 4: paddr=0x00189d3c vaddr=0x40080000 size=0x00400 (  1024) load
I (583) esp_image: segment 5: paddr=0x0018a144 vaddr=0x40080400 size=0x1ae84 (110212) load
I (643) boot: Loaded app from partition at offset 0x10000
I (643) boot: Disabling RNG early entropy source...
I (643) psram: This chip is ESP32-D0WD
I (644) spiram: Found 64MBit SPI RAM device
I (644) spiram: SPI RAM mode: flash 40m sram 40m
I (645) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (646) cpu_start: Pro cpu up.
I (646) cpu_start: Application information:
I (647) cpu_start: Project name:     voicepad_v1
I (647) cpu_start: App version:      1
I (648) cpu_start: Compile time:     Oct 16 2020 13:07:33
I (649) cpu_start: ELF file SHA256:  0c2dc1acebce079c...
I (649) cpu_start: ESP-IDF:          v3.3.2-107-g722043f73-dirty
I (650) cpu_start: Starting app cpu, entry point is 0x4008150c
I (630) cpu_start: App cpu up.
I (1517) spiram: SPI SRAM memory test OK
I (1518) heap_init: Initializing. RAM available for dynamic allocation:
I (1518) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1518) heap_init: At 3FFBB008 len 00024FF8 (147 KiB): DRAM
I (1519) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1520) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1520) heap_init: At 4009B284 len 00004D7C (19 KiB): IRAM
I (1521) cpu_start: Pro cpu start user code
I (1522) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (190) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (192) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (229) VOICEPAD: ADF version is v2.1-77-gebbde85-dirty
I (229) VOICEPAD: [Step 1.0] Create esp_periph_set_handle_t instance and initialize Button
I (230) VOICEPAD: [Step 1.1] Initialize input key callback
I (231) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (232) gpio: GPIO[13]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (233) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (234) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (235) gpio: GPIO[23]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (236) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (238) VOICEPAD: [Step 1.2] Create display service instance
I (239) VOICEPAD: [Step 2.0] Initialize iflyos audio
I (239) A1S: i2c port configured!!!!
W (1241) AC101: reset succeed
I (1250) AC101: init done
I (1250) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1255) AUDIO_THREAD: The media_task task allocate stack on external memory
I (1255) ESP_AUDIO_TASK: media_ctrl_task running...,0x3f8004c4

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                         ESP_AUDIO-v1.6.4-3-g428c5e9                        |
|                     Compile date: Sep 24 2020-15:25:00                     |
------------------------------------------------------------------------------
I (1260) ESP_AUDIO_CTRL: Func:media_ctrl_create, Line:339, MEM Total:4407444 Bytes, Inter:271492 Bytes, Dram:251692 Bytes

I (1261) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (1377) MP3_DECODER: MP3 init
I (1379) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1379) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1382) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (1383) A1S: i2s port configured!!!!
I (1386) AUDIO_WRAPPER: Func:setup_player, Line:158, MEM Total:4394328 Bytes, Inter:255308 Bytes, Dram:235508 Bytes

I (1387) AUDIO_WRAPPER: esp_audio instance is:0x3f8004c4
I (1387) VOICEPAD: [Step 3.0] Initialize iFLYOS service
E (1388) iFLYOS_SERVICE: Func:iflyos_start, Line:424
I (1389) VOICEPAD: [Step 5.0] Initialize recorder engine
I (1390) VOICEPAD: [Step 6.0] Initialize Done

W (1390) I2S: I2S driver already installed
I (1392) A1S: i2s port configured!!!!
task_name   task_state  priority   stack  tasK_num
rec_engine_task R       20      1012    17      1
main            R       1       1364    5       0
IDLE1           R       0       1128    7       1
IDLE0           R       0       1132    6       0
tiT             B       18      2720    12      -1
Tmr Svc         B       5       1620    8       0
iflyos_serv     B       4       7660    16      0
esp_periph      B       5       2224    13      0
ipc1            B       24      584     3       1
ipc0            B       24      476     2       0
media_task      B       6       2224    14      0
player_task     B       1       2648    15      -1
esp_timer       B       22      3684    1       0

I (1394) AUDIO_PIPELINE: link el->rb, el:0x3f8028b4, tag:i2s, rb:0x3f802ca4
I (1397) VOICEPAD: [Step 4.0] Create Wi-Fi service instance
I (1398) AUDIO_PIPELINE: link el->rb, el:0x3f802a14, tag:filter, rb:0x3f804ce4
E (1399) DISPATCHER: exe first list: 0x0
I (1400) AUDIO_ELEMENT: [i2s-0x3f8028b4] Element task created
I (1400) DISPATCHER: dispatcher_event_task is running...
I (1401) AUDIO_THREAD: The filter task allocate stack on external memory
I (1403) AUDIO_ELEMENT: [filter-0x3f802a14] Element task created
I (1403) AUDIO_ELEMENT: [raw-0x3f802b3c] Element task created
I (1404) AUDIO_THREAD: The wifi_serv task allocate stack on external memory
I (1404) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:359, MEM Total:4351760 Bytes, Inter:230912 Bytes, Dram:211112 Bytes

I (1407) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (1407) I2S_STREAM: AUDIO_STREAM_READER,Rate:48000,ch:2
I (1410) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (1413) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1
I (1413) AUDIO_PIPELINE: Pipeline started
I (1413) VOICEPAD: Recorder has been created
I (1414) wifi:wifi driver task: 3ffd4d2c, prio:23, stack:3584, core=0
I (1415) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
Quantized wakeNet5: wakeNet5_v1_hilexin_5_0.95_0.90, mode:0 (Oct 14 2020 16:26:17)
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060f30  A0      : 0x00000000  A1      : 0x3ffcf160
A2      : 0x3f8027ec  A3      : 0x3ffb4b08  A4      : 0x3ffb4b00  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000018  A8      : 0x800feb63  A9      : 0x3ffcf140
A10     : 0x3ffd4f58  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000
A14     : 0x400fe834  A15     : 0x3f413419  SAR     : 0x00000004  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x40098019  LEND    : 0x40098029  LCOUNT  : 0xfffffffd

ELF file SHA256: 0c2dc1acebce079c

Backtrace: 0x00000000:0x3ffcf160 0x7ffffffd:0x3ffcf1d0

Rebooting...`

bulazhang, 2020-10-18:
I try use vscode to build again.

vscode idf extension v0.5.1, IDF v4.1 ADF v2.1. I got many librecorder_engine.a undefined reference.

`[1051/1054] Generating ld/esp32.project.ld
warning: the default selection ESP32_KORVO_DU1906_CODEC_TAS5805M (undefined) of <choice ESP32_KORVO_DU1906_DAC> (defined at E:/Workspace/ESP/esp-adf/components/audio_board/Kconfig.projbuild:27) is not contained in the choice
[1053/1054] Linking CXX executable voicepad_vscode.elf
FAILED: voicepad_vscode.elf
cmd.exe /C "cd . && E:\DevKits\esp-idf-tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\voicepad_vscode.elf.rsp  -o voicepad_vscode.elf  && cd ."
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xdc): undefined reference to `Encoder_Interface_exit'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xe8): undefined reference to `amrwb_enc_header_info'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xec): undefined reference to `get_wakenet_iface'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xf0): undefined reference to `get_wakenet_coeff'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xf4): undefined reference to `vad_create'      
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0xf8): undefined reference to `vad_destroy'     
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0x104): undefined reference to `vad_process'    
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0x108): undefined reference to `Encoder_Interface_Encode'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0x10c): undefined reference to `amrwb_enc_process'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_task+0x110): undefined reference to `amrwb_enc_close'e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_create+0x34): undefined reference to `Encoder_Interface_init'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj):(.literal.rec_engine_create+0x38): undefined reference to `amrwb_enc_open'e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: E:/Workspace/ESP/esp-adf/components/esp-adf-libs/recorder_engine/lib/esp32\librecorder_engine.a(recorder_engine.c.obj): in function `rec_engine_task':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/recorder_engine/recorder_engine.c:659: undefined reference to `Encoder_Interface_exit'
[... cut ...]
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/recorder_engine/recorder_engine.c:707: undefined reference to `Encoder_Interface_init'
e:/devkits/esp-idf-tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/recorder_engine/recorder_engine.c:718: undefined reference to `amrwb_enc_open'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake --build .”已终止，退出代码: 1。`

bulazhang, 2020-10-23:
After a day of analysis，I found the panic generated by rec_engine_create API.

As DuerOS demo, the environments are IDF 4.1, ADF 2.1, xtensa-esp32-elf/esp-2020r3-8.4.0

`rec_config_t eng = DEFAULT_REC_ENGINE_CONFIG();
    eng.vad_off_delay_ms = 800;
    eng.wakeup_time_ms = 10 * 1000;
    eng.evt_cb = rec_engine_cb;
#ifdef CONFIG_ESP_LYRAT_MINI_V1_1_BOARD
    eng.open = recorder_pipeline_open_for_mini;
#else
    eng.open = recorder_pipeline_open;
#endif
    eng.close = recorder_pipeline_close;
    eng.fetch = recorder_pipeline_read;
    eng.extension = NULL;
    eng.support_encoding = false;
    eng.user_data = NULL;
    rec_engine_create(&eng); `

Guru Meditation Error: Core 1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC : 0x00000000 PS : 0x00060f30 A0 : 0x00000000 A1 : 0x3ffcf160
A2 : 0x3f8027ec A3 : 0x3ffb4b08 A4 : 0x3ffb4b00 A5 : 0x00000000
A6 : 0x00000000 A7 : 0x00000018 A8 : 0x800feb63 A9 : 0x3ffcf140
A10 : 0x3ffd4f58 A11 : 0x00000000 A12 : 0x00000000 A13 : 0x00000000
A14 : 0x400fe834 A15 : 0x3f413419 SAR : 0x00000004 EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000 LBEG : 0x40098019 LEND : 0x40098029 LCOUNT : 0xfffffffd

jason-mao, 2020-10-27:
@bulazhang Thanks for your report. The fix is coming.

closed 2020-10-27 by espressif-bot