espressif/esp-adf #480: Building  errors  in Windows CMD environment

Patsjemoe, 2020-10-16:
Hi, 
Use ESP-ADF-V2.1, ESP-IDF v 4.1 unders Window CMDs, to compile and load VOIP example.
IDF.py menuconfig = ok
IDF.py build gives strange errors, related to parts I do not use in the VOIP application.

C:/path/esp/esp-adf 2.1/components/display_service/led_bar/led_bar_ws2812.c:27:10: fatal error: board.h: No such file or directory
 #include "board.h" 

C:/path/esp/esp-adf-v2.1/components/display_service/display_service.c:34:10: fatal error: board.h: No such file or directory
 #include "board.h"

...+ 7 more errors, always referring to the board.h file.

thanks in advance.
regards
Ludo

Patsjemoe, 2020-10-18:
no feedback, closed without solution

closed 2020-10-18 by Patsjemoe

jalakamkiranadspl, 2021-03-24:
I am getting the same error with the ubuntu os on esp32 audio kit from ai thinker. Any solutions?