espressif/esp-adf #484: ESP-ADF Audio resample max_indata_bytes ignored

ypiotrp, 2020-10-20:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version: v4.1-1-gb16edf06f-dirty with patch from https://github.com/espressif/esp-idf/issues/4967#issuecomment-707520930_
- ADF version: v2.1 and newest master
- Build System: make
- Compiler version: xtensa-esp32-elf-gcc 8.2.0 2020r2
- Operating System: win10
- Using an IDE?: No
- Power Supply: USB

## Problem Description
What I want to do is resample raw audio data from WAV file. The original sample rate is 22050 Hz and I want to get 8000 Hz. I use for this `esp_resample_create()` and `esp_resample_run`.

Is this a proper way to use the resample feature? How can I use I use it on raw samples from buffer?

## Expected Behavior
I expected to get resampled audio signal in my `output_buffer`.

## Actual Behavior
I get the following error several times and then the ESP32 reboots.
```FIR_RESAMPLE: inbuf length greater than max_indata_bytes. max_indata_bytes 1 inbuf length 512```

The `max_indata_bytes` is set to 512 in `resample_info_t resample_config`. I don't know from the `esp_resample_run()` function gets the 1 value.

## Steps to reproduce
Use the code I pasted below. 

## Code to reproduce this issue
```
uint32_t samples_to_push = original_44100hz_mono_1s_wav_len;
    unsigned char output_buffer[512], input_buffer[512];
    int i=0;
    int samples_written;
    void* resample_handler;
    resample_info_t resample_config = {
        .src_rate           = 44100,
        .src_ch             = 1,
        .dest_rate          = 8000,
        .dest_ch            = 1,
        .mode               = RESAMPLE_DECODE_MODE,
        .sample_bits        = 16,
        .max_indata_bytes   = 512,
        .type               = ESP_RESAMPLE_TYPE_RESAMPLE,
        .complexity         = 2,
    };

    while (samples_to_push > 512) {
        memcpy(input_buffer, &original_44100hz_mono_1s_wav[44+i*512], 512);
        resample_handler = esp_resample_create(&resample_config, (unsigned char **)input_buffer, (unsigned char **)output_buffer);
        esp_resample_run(&resample_handler, &resample_config, input_buffer, output_buffer, 512, &samples_written);
        for(int k=0; k<samples_written; k++) {
            printf("%d, ", output_buffer[k]);
        }
        samples_to_push -= samples_written;
    }
```
where `original_44100hz_mono_1s_wav` is a char buffer with WAV data included as a header.

houhaiyan, 2020-10-21:
There has a wrong use for those functions.
1. i++ must exist in while cycle
2. Is sample rate of the source audio stream  22050Hz or 44100Hz?
3. For one audio stream, the esp_resample_create is needed use one time.  refer in `esp_rasample.h`

ypiotrp, 2020-10-21:
Hi @houhaiyan, thank you for your suggestions.
1. You are right, I forgot to increment the i variable in the while cycle.
2. As for the sampling rate of the audio stream I checked it again and it is 44100Hz. 
3. I moved the `esp_resample_create()` outside the while loop. 
After that I flashed the ESP again and the problem remains. I still get the error `FIR_RESAMPLE: inbuf length greater than max_indata_bytes. max_indata_bytes 1 inbuf length 512
` after running the `esp_resample_run()` in the while loop. I checked if the `max_indata_bytes` value from the config somehow changes but it doesn't - always getting 512. Where does the 1 in the error come from?

houhaiyan, 2020-10-21:
@ypiotrp  please check you the number channel of audio stream. max_indata_bytes =  the value you set *  src_ch.
And you can refer in `filter_resample.c`. I think the file will help you.

houhaiyan, 2020-10-22:
@ypiotrp How's about your feature?

ypiotrp, 2020-10-22:
Hi @houhaiyan. Thank you for your help 🙂 I followed `filter_resample.c` and everything works right now. The issue can be closed.

closed 2020-10-23 by jason-mao