espressif/esp-adf #485: Aux input only redirect to speakers when sound card initialized with AUDIO_HAL_CODEC_MODE_LINE_IN

jujax, 2020-10-22:
Hello,

I try to use my ESP32-LyraT 4.3 to send sound to Bluetooth headset from line input.
Actualy ESP-ADF library enable line input only when sound card is initialized with :
```c
audio_hal_ctrl_codec(board_handle->audio_hal, AUDIO_HAL_CODEC_MODE_LINE_IN, AUDIO_HAL_CTRL_START);
```
with this mode, sound is redirected only to speakers or line output. Pipeline doesn't send sound to writer streams.

To bypass this limitation I use directly ES8388 driver library with this function located in ```es8388.h``` :
```c
es8388_write_reg(ES8388_ADCCONTROL2, ADC_INPUT_LINPUT2_RINPUT2);
```
By the way, sound is really dirty without setting PGA gain to 0db :
```c
es8388_write_reg(ES8388_ADCCONTROL1, 0x00); // value is 0xbb when card is init
```
It works very well with these modifications, and I create a [github deposit](https://github.com/jujax/lyrat_passthru) to make an example.
Maybe I disrespect some coding and github conventions (bad licensing for code, bad README.md), every improvements are welcome.
And sorry for my english.

Gaochenchang, 2020-10-27:
If line_ in is used as input, setting the gain to 0 is correct.

jason-mao, 2022-02-08:
This topic has become inactive so I'm going to close the issue. Please reopen this if you have any questions or need any further assistance.

TAMHAN, 2023-11-17:
The man is right. His error description is correct, and it needs to be fixed.

TAMHAN, 2023-11-17:
Please reopen this issue as it is valid and prudent

jason-mao, 2024-04-07:
As @TAMHAN suggested, reopen it.

TAMHAN, 2024-04-07:
Thank you, @jason-mao

t36512, 2024-04-21:
Hi,
Thanks to @jujax 's exmple project I got my Lyrat 4.3 board working as well. The only issue in my case is extremely low volume at the paired bluetooth speaker. I spend hours looking for solutions with no luck at all. Would be great if someone can provide me some hint to fix it. Any suggestion is highly appreicated.