espressif/esp-adf #486: DLNA Example: [2/3] Linking CXX executable dlna_example.elf

SeeOneeee, 2020-10-23:
So tried to fix this problem for a few hours. No fixes worked for me so far. 
I get this error: [2/3] Linking CXX executable dlna_example.elf
Everything is updated and i reinstalled ESP-IDF and ESP-ADF. 
I didn't modify anything in the project.
I had problems building other projects in the ADF and I used this fix: https://github.com/espressif/esp-adf/issues/438#issuecomment-667678452

jason-mao, 2020-10-27:
@SeeOneeee Please provide the error log

SeeOneeee, 2020-10-28:
@jason-mao Here is the complete Error Log - it is a little bit long.

`Executing action: all (aliases: build)
Running ninja in directory /home/User/esp/esp-adf/examples/advanced_examples/dlna/build
Executing "ninja all"...
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable dlna_example.elf
FAILED: dlna_example.elf 
: && /home/User/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -mlongcalls -Wno-frame-address  CMakeFiles/dlna_example.elf.dir/project_elf_src.c.obj CMakeFiles/dlna_example.elf.dir/rootDesc.xml.S.obj CMakeFiles/dlna_example.elf.dir/rootDescLogo.xml.S.obj CMakeFiles/dlna_example.elf.dir/rootDescService.xml.S.obj CMakeFiles/dlna_example.elf.dir/SoapResponseErr.xml.S.obj CMakeFiles/dlna_example.elf.dir/SoapResponseOk.xml.S.obj CMakeFiles/dlna_example.elf.dir/devicelist.xml.S.obj -o dlna_example.elf  esp-idf/xtensa/libxtensa.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_ipc/libesp_ipc.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/soc/libsoc.a  esp-idf/vfs/libvfs.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/tcpip_adapter/libtcpip_adapter.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/esp_event/libesp_event.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/lwip/liblwip.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/driver/libdriver.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/perfmon/libperfmon.a  esp-idf/esp32/libesp32.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/app_trace/libapp_trace.a  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/coap/libcoap.a  esp-idf/console/libconsole.a  esp-idf/nghttp/libnghttp.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc_cal/libesp_adc_cal.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/mdns/libmdns.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_serial_slave_link/libesp_serial_slave_link.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/jsmn/libjsmn.a  esp-idf/json/libjson.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/spiffs/libspiffs.a  esp-idf/ulp/libulp.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/audio_sal/libaudio_sal.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/display_service/libdisplay_service.a  esp-idf/esp_dispatcher/libesp_dispatcher.a  esp-idf/audio_hal/libaudio_hal.a  esp-idf/esp-sr/libesp-sr.a  esp-idf/esp-adf-libs/libesp-adf-libs.a  esp-idf/audio_pipeline/libaudio_pipeline.a  esp-idf/esp_peripherals/libesp_peripherals.a  esp-idf/audio_board/libaudio_board.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/dueros_service/libdueros_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/playlist/libplaylist.a  esp-idf/esp_dlna/libesp_dlna.a  esp-idf/main/libmain.a  -Wl,--cref -Wl,--Map=/home/User/esp/esp-adf/examples/advanced_examples/dlna/build/dlna_example.map  -fno-rtti  -fno-lto  esp-idf/asio/libasio.a  esp-idf/cbor/libcbor.a  esp-idf/cmock/libcmock.a  esp-idf/unity/libunity.a  esp-idf/coap/libcoap.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/esp_websocket_client/libesp_websocket_client.a  esp-idf/expat/libexpat.a  esp-idf/freemodbus/libfreemodbus.a  esp-idf/libsodium/liblibsodium.a  esp-idf/mqtt/libmqtt.a  esp-idf/openssl/libopenssl.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/protocomm/libprotocomm.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/mdns/libmdns.a  esp-idf/json/libjson.a  esp-idf/adf_utils/libadf_utils.a  esp-idf/jsmn/libjsmn.a  esp-idf/battery_service/libbattery_service.a  esp-idf/bluetooth_service/libbluetooth_service.a  esp-idf/coredump_upload_service/libcoredump_upload_service.a  esp-idf/esp_event_cast/libesp_event_cast.a  esp-idf/input_key_service/libinput_key_service.a  esp-idf/ota_service/libota_service.a  esp-idf/audio_stream/libaudio_stream.a  esp-idf/tone_partition/libtone_partition.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  esp-idf/wifi_service/libwifi_service.a  esp-idf/esp_actions/libesp_actions.a  -L/home/User/esp/esp-adf/components/wifi_service/airkiss_config  -l [... cut ...]
/home/User/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/User/esp/esp-adf/components/esp-adf-libs/esp_audio/lib/esp32/libesp_audio.a(media_ctrl.c.obj):/Users/maojianxin/duros/esp-adf-libs-source/esp_audio/src/media_ctrl.c:507: more undefined references to `xQueueGenericReceive' follow
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

SeeOneeee, 2020-10-28:
@jason-mao Thanks for the reply! Here is the Error Log - it is quite long.
[Error_Log.txt](https://github.com/espressif/esp-adf/files/5451897/Error_Log.txt)

jason-mao, 2020-12-02:
@SeeOneeee Would you please test your problem with latest master ?

grewhit25, 2020-12-02:
I have tested this against the latest master and it is still failing for me; error log same as @jason-mao

jason-mao, 2020-12-02:
What IDF version you use?

grewhit25, 2020-12-02:
@jason-mao my idf.py --version
ESP-IDF v4.3-dev-1901-g178b122c1

jason-mao, 2020-12-02:
BTW, did you test with other versions, such as v4.2 v4.1.

grewhit25, 2020-12-02:
Yes, my bad.
I can confirm that I rolled back to v4.1 and v4.2-rc; both compiled the example successfully.
Apologies for time wasting.

closed 2022-02-08 by jason-mao