espressif/esp-adf #488: HFP 听不清来自phone的声音

bigfeng12, 2020-10-27:
- 测试环境：
ESP32-LyraTD-MSC V2.2

- 测试步骤

1. 编译烧写a2dp_sink_and_hfp example（examples\get-started\pipeline_a2dp_sink_and_hfp）

2. phone链接ESP-ADF-AUDIO，耳机插入msc v2.2的耳机口

3. phone播放音乐，耳机可以听到音乐（因为msc v2.2 board的dsp不支持44.1K sample rate，需要寄一个44.1 to 48的resample filter）

4. phone拨打10086，耳机里名听不清电话声音，问题复现。

- 原因分析：

从代码来看，HFP link建立之后，强制把sample rate有44.1K变到8K，channel有2变成1，

但是上述第三条里的resample filter没有改变导致。

我尝试在bt_app_hf_client_audio_open中用rsp_filter_set_src_info(filter, 8000, 1)重启filter，但是过一会board 重启：

E (19003) BT_HF: APP HFP event: BATTERY_LEVEL_IND_EVT
E (19013) BT_HF: --battery level 0
W (19133) BT_BTM: SCO xmit Q overflow, pkt dropped
W (19133) BT_BTM: stat 10
W (19263) BT_BTM: SCO xmit Q overflow, pkt dropped
W (19263) BT_BTM: stat 10
W (19573) BT_BTM: SCO xmit Q overflow, pkt dropped
W (19573) BT_BTM: stat 10
W (19633) BT_BTM: SCO xmit Q overflow, pkt dropped
W (19633) BT_BTM: stat 10
W (19893) BT_BTM: SCO xmit Q overflow, pkt dropped
W (19893) BT_BTM: stat 10
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40031b6f  PS      : 0x00060231  A0      : 0x800837ad  A1      : 0x3ffbe2a0
A2      : 0x3ffb142c  A3      : 0x00000000  A4      : 0x3ffae70c  A5      : 0x00000000
A6      : 0x3ffb8d40  A7      : 0x40083604  A8      : 0x80031b58  A9      : 0x3ffbe280
0x40083604: r_ea_alarm_set at ??:?

A10     : 0x00016460  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd06b0
A14     : 0x00000007  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000008c  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 0 was running in ISR context:
EPC1    : 0x40031b6f  EPC2    : 0x4017bdd2  EPC3    : 0x00000000  EPC4    : 0x00000000

- 问题解决：

看看esp是否可以各一个example，来动态改变resample的参数？或者有其他的解决方案on msc v2.2 board？

espressif-zhanghu, 2020-10-27:
@bigfeng12  
1. add “i2s_stream_set_clk(i2s_stream_reader, 8000, 16, 1);”  for every board  before  “i2s_stream_set_clk(i2s_stream_writer, 8000, 16, 1);”
2. set "BTM_SCO_XMIT_QUEUE_THRS 50" in components/bt/bluedroid/stack/btm/include/btm_int.h

bigfeng12, 2020-10-28:
> @bigfeng12
> 
> 1. add “i2s_stream_set_clk(i2s_stream_reader, 8000, 16, 1);”  for every board  before  “i2s_stream_set_clk(i2s_stream_writer, 8000, 16, 1);”
> 2. set "BTM_SCO_XMIT_QUEUE_THRS 50" in components/bt/bluedroid/stack/btm/include/btm_int.h

@espressif-zhanghu 

pipeline_a2dp_sink_and_hfp的代码中没有i2s_stream_set_clk(i2s_stream_writer, 8000, 16, 1);，我的修改代码如下，并不能fix问题：

`static void bt_app_hf_client_audio_open(void)
{
    ESP_LOGE(BT_HF_TAG, "bt_app_hf_client_audio_open");
    int sample_rate = 8000;
    audio_element_info_t bt_info = {0};
    audio_element_getinfo(bt_stream_reader, &bt_info);
    bt_info.sample_rates = sample_rate;
    bt_info.channels = 1;
    bt_info.bits = 16;
    audio_element_setinfo(bt_stream_reader, &bt_info);
    audio_element_report_info(bt_stream_reader);
    ++i2s_stream_set_clk(i2s_stream_writer, 8000, 16, 1);
    ++i2s_stream_set_clk(i2s_stream_reader, 8000, 16, 1);
    ++rsp_filter_set_src_info(filter, 8000, 1);
}
`

closed 2020-12-28 by espressif-bot