espressif/esp-adf #494: algo_stream error

utkutpcgl, 2020-11-05:
I guess I faced a bug. I would be glad if you could help.

When I tried to build the project with algo stream added I face the following error:

`[5/7] Generating ld/esp32.project.ld
warning: the default selection ESP32_KORVO_DU1906_CODEC_TAS5805M (undefined) of <choice ESP32_KORVO_DU1906_DAC> (defined at C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/audio_board/Kconfig.projbuild:27) is not contained in the choice
[6/7] Linking CXX executable dinleme_deneme.elf
FAILED: dinleme_deneme.elf
cmd.exe /C "cd . && C:\Users\utkum\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\dinleme_deneme.elf.rsp  -o dinleme_deneme.elf  && cd ."
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x48): undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x4c): undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x58): undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_noisemean_cal':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:123: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:131: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_pro':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:217: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:257: undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `esp_ns_alloc':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.h:30: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/Documents/esp32_idf4.1_latestADF/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_init':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:295: undefined reference to `esp_kiss_fftr_alloc'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

As far as I understand libesp_audio_processor.a has a reference to **esp_kiss_fftr** in _noise suppression related functions_ in the **ns_core** file.

I use:

> Latest adf version
> idf version 4.1
> Windows idf command prompt
> esp32_lyrat board.

Have a nice day.

_Originally posted by @utkutpcgl in https://github.com/espressif/esp-adf/issues/464#issuecomment-715164315_

Patsjemoe, 2020-11-08:
Hi,
@utkutpcgl, did you get the example code already? 
Do you combine algorithm stream with pipeline?
Regards

utkutpcgl, 2020-11-11:
Hi,
I could not find the example yet. Yes, I think we have to combine it with the pipeline as an element.
Best wishes.

utkutpcgl, 2020-11-25:
The example is available.

utkutpcgl, 2020-11-25:
I tried to flash LyraT with the structure given in the example code again, however, there is still the same error:

>`c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x3c): undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x40): undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o):(.literal.ns_pro+0x4c): undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_noisemean_cal':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:123: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:131: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_pro':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:217: undefined reference to `esp_kiss_fftr'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:257: undefined reference to `esp_kiss_fftri'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `esp_ns_alloc':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.h:30: undefined reference to `esp_kiss_fftr_alloc'
c:/users/utkum/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/utkum/OneDrive/Belgeler/esp/esp-adf/components/esp-sr/acoustic_algorithm\libesp_audio_processor.a(ns_core.o): in function `ns_init':
/home/zhaoyang/esp/esp_sr_lib/components/esp_audio_processor/noise_suppression/ns_core.c:295: undefined reference to `esp_kiss_fftr_alloc'`

It has been 20 days since I reported this error, so, it would be nice at least to see a response. 
Thanks in advance.

jason-mao, 2020-11-26:
@utkutpcgl Sorry for missing, would you please provide your IDF version and ADF commit or version?

utkutpcgl, 2020-11-26:
I use latest esp-adf and esp-idf 4.1. Thanks @jason-mao , I understand you have not much time.

jason-mao, 2020-11-26:
@utkutpcgl  There is a quick fixes as follow.
```c
cd $ADF_PATH/components/esp-sr/
git checkout master
git pull
```
Then modify the CMakeLists.txt, Add `add_prebuilt_library(esp_audio_processor "${CMAKE_CURRENT_SOURCE_DIR}/acoustic_algorithm/libesp_audio_processor.a" PRIV_REQUIRES esp-sr)` behind  
https://github.com/espressif/esp-sr/blob/eec128886f882f4bd446eb53588c16858a68acb1/CMakeLists.txt#L23.
,such as 
[image]

utkutpcgl, 2020-11-26:
Thanks for the fast response, I will share the results.

utkutpcgl, 2020-12-08:
It works fine as far as I have tested. Thanks!

closed 2020-12-08 by utkutpcgl

Patsjemoe, 2020-12-11:
Hi,
Did you use it as it is, or did you make any changes?
I am trying to implement it in VOIP, but am still struggling due to
unclarities in description of multi input/output ringbuffers
regards
Ludo