espressif/esp-adf #501: audio_hal_set_mute() not working or misconfigured

bs-eng, 2020-11-11:
Hi!

I am using the LyraT v4.3 hardware to play audio files from sd-card. For this I have set up an audio pipeline (like in example play_sdcard_mp3_control_example.c).
This works fine so far.

While no audio is played I noticed an audible level of noise on the speaker. Since the codec and the power amplifier on the LyraT board support mute I looked for a mute function and found
`esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);`
It seems to control the enable pin of the power amplifier to achieve a real mute. For the LyraTV4.3 that would be GPIO21, which is connected to the PA enable.

The function audio_hal_set_mute() has no effect (measured voltage on pin).
From the log I see that GPIO21 seems to be misconfigured. The configuration is done somewhere in the audio pipeline init. The example code had no extra init procedure for GPIO21.

What is wrong with the function audio_hal_set_mute() or the GPIO init in audio module?
How can I fix this?

Thanks a lot!
Cheers!

I (586) AUDIOTASK: [ 2 ] Start codec chip
I (586) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
**E (586) gpio: gpio_install_isr_service(412): GPIO isr service already installed**
I (606) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (606) ES8388_DRIVER: init,out:02, in:00
I (616) AUDIO_HAL: Codec mode is 2, Ctrl:1
I (616) AUDIOTASK: [ 3 ] Create and start input key service
I (616) AUDIOTASK: [4.0] Create audio pipeline for playback
I (626) AUDIOTASK: [4.1] Create i2s stream to write data to codec chip
I (636) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (636) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (646) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (666) LYRAT_V4_3: I2S0, MCLK output by GPIO0
I (666) AUDIOTASK: [4.2] Create mp3 decoder to decode mp3 file
I (676) MP3_DECODER: MP3 init
I (676) AUDIOTASK: [4.3] Create resample filter
I (686) AUDIOTASK: [4.4] Create fatfs stream to read data from sdcard
I (686) AUDIOTASK: [4.5] Register all elements to audio pipeline
I (696) AUDIOTASK: [4.6] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->resample-->i2s_stream-->[codec_chip]
I (706) AUDIO_PIPELINE: link el->rb, el:0x3ffd0010, tag:file, rb:0x3ffd03b0
I (716) AUDIO_PIPELINE: link el->rb, el:0x3ffcf94c, tag:mp3, rb:0x3ffd24ec
I (726) AUDIO_PIPELINE: link el->rb, el:0x3ffcfcc4, tag:filter, rb:0x3ffd2e28
I (726) AUDIOTASK: [5.0] Set up  event listener
I (736) AUDIOTASK: [5.1] Listen for all pipeline events

Gaochenchang, 2020-11-26:
Hi, you use `esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);`  can achieve the effect of mute. If you want no sound output, you can use `esp_err_t gpio_set_direction(gpio_num_t gpio_num, gpio_mode_t mode);` turns off the output function of GPIO21.
Best Wishes.

bs-eng, 2020-12-09:
Dear Gaochenchang,

thx for the proposed solutions.
The function `esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);` does not work properly due to conflict on ADF components. Please see my original post, log entry at (586). There seems to be a conflict between isr routines.

The use of `esp_err_t gpio_set_direction(gpio_num_t gpio_num, gpio_mode_t mode);` is a workaround to me. It does the job, but again it is in conflict with isr routines and it breaks the ADF abstraction layers.

Can you advise on when this conflict in ADF components will be fixed?

Thanks a lot!
Cheers
BS

Gaochenchang, 2020-12-14:
`E (586) gpio: gpio_install_isr_service(412): GPIO isr service already installed ` is normal. You can ignore this log and use `esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);` needs to contain es8388. h.

bs-eng, 2020-12-16:
Dear Gaochenchang,

maybe I am not clear enough... :-)

The function `esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);` has **no** effect. It does **not** mute the output.

I followed your advice to also include es8388.h. That made no difference.
I am missing something else or is it a bug in the function?

Thanks a lot!
BS

Gaochenchang, 2020-12-23:
Dear bs-eng,

I think you mean removing the noise floor. Under normal circumstances, `esp_err_t audio_hal_set_mute(audio_hal_handle_t audio_hal, bool mute);` can achieve this effect, and there is another way as you said, not to power the PA.

closed 2023-09-05 by jason-mao