espressif/esp-adf #505: AMRW causes watchdog error.

utkutpcgl, 2020-11-13:
I use amrwb in my pipeline and when it starts the watchog timer raises an error repeatedly that it was not reset. I use Lyrat, espidf 4.1 and latest adf with esp idf on CMD. Thanks in advance.

jason-mao, 2020-11-25:
Please use `audio_sys_get_real_time_stats` to make cpu loading in balance.

utkutpcgl, 2020-11-25:
Thanks, I will share the results later.

utkutpcgl, 2020-11-25:
I assigned core 0 to the amrwb encoder (works faster this way) and core1 to amrwb decoder as task cores (task_core variable of config) and now it works without the watchdog timer errors. Thanks! I see that audio_sys_get_real_time_stats is new. The library is progressing well. Thanks again.

closed 2020-11-25 by utkutpcgl

Patsjemoe, 2020-11-25:
Hi,
Can you share how you assigned the encoder and decoder to core 0 and 1?
Maybe this will help me to  solve my problem with voip aec algorithm
stream...

Thanks
Ludo

utkutpcgl, 2020-11-26:
@Patsjemoe I changed the task core of amr decoder by modifying the following struct: 
[image]
I guess your problem was a little different, but hopefully, it helps. 
Best wishes.