espressif/esp-adf #509: undefined reference to `ogg_stream_packetout'

liangzhu2008, 2020-11-19:
我尝试编译 `esp-adf\examples\cli`  这个例子，得到这个结果：

```none
[2/6] Performing build step for 'bootloader'
ninja: no work to do.
[3/4] Linking CXX executable cli_app.elf
FAILED: cli_app.elf
cmd.exe /C "cd . && C:\Users\zhu\.espressif\tools\xtensa-esp32-elf\esp-2020r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\cli_app.elf.rsp  -o cli_app.elf  && cd ."
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_stream_packetout+0x0): undefined reference to `ogg_stream_packetout'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_stream_pagein+0x0): undefined reference to `ogg_stream_pagein'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_page_granulepos+0x0): undefined reference to `ogg_page_granulepos'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_page_serialno+0x0): undefined reference to `ogg_page_serialno'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_stream_init+0x0): undefined reference to `ogg_stream_init'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_sync_wrote+0x0): undefined reference to `ogg_sync_wrote'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_sync_buffer+0x0): undefined reference to `ogg_sync_buffer'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_sync_pageout+0x0): undefined reference to `ogg_sync_pageout'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_sync_clear+0x0): undefined reference to `ogg_sync_clear'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_stream_clear+0x0): undefined reference to `ogg_stream_clear'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_sync_init+0x0): undefined reference to `ogg_sync_init'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj):(.literal.opus_stream_reset_serialno+0x0): undefined reference to `ogg_stream_reset_serialno'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_stream_packetout':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:129: undefined reference to `ogg_stream_packetout'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_stream_pagein':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:134: undefined reference to `ogg_stream_pagein'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_page_granulepos':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:139: undefined reference to `ogg_page_granulepos'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_page_serialno':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:143: undefined reference to `ogg_page_serialno'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_stream_init':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:148: undefined reference to `ogg_stream_init'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_sync_wrote':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:153: undefined reference to `ogg_sync_wrote'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_sync_buffer':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:158: undefined reference to `ogg_sync_buffer'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_sync_pageout':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:163: undefined reference to `ogg_sync_pageout'
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_sync_clear':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:168: undefined reference to `ogg_sync_clear'
[... cut ...]
c:/users/zhu/.espressif/tools/xtensa-esp32-elf/esp-2020r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Work/ESP32/esp-adf/components/esp-adf-libs/esp_codec/lib/esp32\libesp-opus.a(opus_interface.c.obj): in function `opus_stream_reset_serialno':
/home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-opus/opus_interface.c:183: undefined reference to `ogg_stream_reset_serialno'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

jason-mao, 2020-11-19:
@liangzhu2008 帮提供一下你的 IDF 版本和ADF commit

liangzhu2008, 2020-11-23:
ESP-IDF: 4.1

```none
commit 5ef1b390026270503634ac3ec9f1ec2e364e23b2 (HEAD, tag: v4.1)
Merge: bd72a9ab2 be1351337
Author: Angus Gratton <angus@espressif.com>
Date:   Thu Aug 13 07:54:36 2020 +0800

    Merge branch 'fix/prov_ble_read_offset_v4.1' into 'release/v4.1'

    BLE provisioning: Add check for valid ble read offset (v4.1)

    See merge request espressif/esp-idf!9745
```

ESP-ADF:

```none
commit 6016cd27b0cc455c7dd16a8400afead17fa2930f (HEAD -> master, origin/master, origin/HEAD)
Merge: fa47509 8e1328b
Author: Jason-mao <maojianxin@espressif.com>
Date:   Tue Oct 27 19:39:53 2020 +0800

    Merge branch 'feature/add_lightduer_v3.1.1' into 'master'

    Add lightduer v3.1.1

    See merge request adf/esp-adf-internal!650
```

jason-mao, 2022-02-08:
This issue has been fixed on latest master.

closed 2022-02-08 by jason-mao