espressif/esp-adf #511: Is it possible to stream through http and save audio to the sd at the same time?

Saynok, 2020-11-23:
Hello, I'm doing cloud speech recognition implementation on esp32 LyraT v3.4 board, and I'm facing to save audio track to the sd card for debug.

- Is it good idea to make them work together?
- Could you please give me advice how to make them work correctly and clear? 
- do you have some guidelines for a pipelines setup and using?
- what's threading rules need to keep? my application is using different threads, so, I got some errors while calling pipeline methods working on different core.

There is a lot of questions, I would be really appreciate to got any comments and answers from community.

Saynok, 2020-11-23:
There is another developer (AUD-2533) talking about this, please let us know if some wiki or guideline could be created for us

jason-mao, 2020-11-25:
@Saynok Is that you want to download HTTP music to playing and saving to sdcard?

Saynok, 2020-11-25:
Hi! No, I need to record audio from microphone (I2S) and stream it into the Google Speech to Text Api and to the ad card in parallel

Saynok, 2020-11-25:
I got some results with making two pipelines together. Each of them has own audio_elements, but I got problem that I got half of the file on sd card and probably half of stream sent to the google STT cloud, because it stops recognising. 

I think 2 pipelines can't use one source in one time, my error means they just divide bytes from i2s. How to avoid that?

jason-mao, 2020-12-02:
@Saynok Actully, I have not got your exactly  problem from your description. But  Iooks like you are trying and almost success. I provide more explains hope to help you .https://github.com/espressif/esp-adf/blob/73a2ec52436ce30195c010c05c9382bfac09eadc/components/audio_stream/i2s_stream.c#L269
 the i2s_stream will copy the read mics data to multi ring buffer. If you suspect  `half of the file ` it's missing data, you can not use multiple output, just write data to scared before sent to google.

closed 2021-09-26 by jason-mao