espressif/esp-adf #512: Combining wakenet with recorder

rjam-xav, 2020-11-23:
Hello,

I am trying to test a basic functionality by integrating the examples from asr and recorder to use hi Jeson to wakeup and then recording. My code is below:. I am able to iteratively listen and have the recording countdown begin but then the REC file is generated though it is unreadable. Since I can iterate over the wakenet and recording without error, I am unsure where the error is coming from? is the pipeline not proceeding to the wav encoding and fstream write? if so why is that the case?

`/* Examples of speech recognition with multiple keywords.

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"
#include "board.h"
#include "audio_pipeline.h"
#include "fatfs_stream.h"
#include "i2s_stream.h"
#include "raw_stream.h"
#include "esp_audio.h"
#include "esp_wn_iface.h"
#include "esp_wn_models.h"
#include "esp_mn_iface.h"
#include "esp_mn_models.h"
#include "mp3_decoder.h"
#include "filter_resample.h"
#include "rec_eng_helper.h"

#include <string.h>
#include "sdkconfig.h"
#include "audio_element.h"
#include "audio_event_iface.h"
#include "audio_common.h"
#include "fatfs_stream.h"
#include "wav_encoder.h"
#include "esp_peripherals.h"
#include "periph_sdcard.h"

static const char* TAG = "example_asr_keywords";
#define RECORD_TIME_SECONDS (5)

typedef enum {
    WAKE_UP = 1,
} asr_wakenet_event_t;

typedef enum {
    ID0_DAKAIKONGTIAO = 0,
    ID1_GUANBIKONGTIAO = 1,
    ID2_ZENGDAFENGSU = 2,
    ID3_JIANXIOAFENGSU = 3,
    ID4_SHENGGAOYIDU = 4,
    ID5_JIANGDIYIDU = 5,
    ID6_ZHIREMOSHI = 6,
    ID7_ZHILENGMOSHI = 7,
    ID8_SONGFENGMOSHI = 8,
    ID9_JIENENGMOSHI = 9,
    ID10_GUANBIJIENENGMOSHI = 10,
    ID11_CHUSHIMOSHI = 11,
    ID12_GUANBICHUSHIMOSHI = 12,
    ID13_DAKAILANYA = 13,
    ID14_GUANBILANYA = 14,
    ID15_BOFANGGEQU = 15,
    ID16_ZANTINGBOFANG = 16,
    ID17_DINGSHIYIXIAOSHI = 17,
    ID18_DAKAIDIANDENG = 18,
    ID19_GUANBIDIANDENG = 19,
    ID_MAX,
} asr_multinet_event_t;
[... 320 lines cut ...]
            break;
        case ID15_BOFANGGEQU:
            ESP_LOGI(TAG, "turn on");
            break;
        case ID16_ZANTINGBOFANG:
            ESP_LOGI(TAG, "turn off");
            break;
        case ID17_DINGSHIYIXIAOSHI:
            ESP_LOGI(TAG, "timer one hour");
            break;
        case ID18_DAKAIDIANDENG:
            ESP_LOGI(TAG, "turn on lignt");
            break;
        case ID19_GUANBIDIANDENG:
            ESP_LOGI(TAG, "turn off lignt");
            break;
        default:
            ESP_LOGI(TAG, "not supportint mode");
            break;
        }
        return ESP_OK;
    }
    return ESP_FAIL;
}
`

rjam-xav, 2020-11-23:
update: the REC file created is empty

Saynok, 2020-11-23:
wow, I had same problem some days ago, and asked about this [https://github.com/espressif/esp-adf/issues/511](issue thread)

The key to achieve the correct pipeline working - to have separate **pipelines** for each process, separate **audio_elements**, 

like this `void system_init_fatfs()
{
    ESP_LOGI(TAG_SYSTEM, "fatfs initialisation");
    audio_board_sdcard_init(periph_handle, SD_MODE_1_LINE);

    fatfs_stream_cfg_t fatfs_cfg_reader = FATFS_STREAM_CFG_DEFAULT();
    fatfs_cfg_reader.task_prio = 8;

    fatfs_cfg_reader.type = AUDIO_STREAM_READER;
    fatfs_stream_reader = fatfs_stream_init(&fatfs_cfg_reader);

    fatfs_stream_cfg_t fatfs_cfg_writer = FATFS_STREAM_CFG_DEFAULT();
    fatfs_cfg_writer.task_prio = 8;

    fatfs_cfg_writer.type = AUDIO_STREAM_WRITER;
    fatfs_stream_writer = fatfs_stream_init(&fatfs_cfg_writer);
}`

and this `audio_pipeline_handle_t pipeline_i2s_to_uSD;
audio_pipeline_handle_t pipeline_uSD_to_i2s;
audio_pipeline_handle_t pipeline_i2s_to_http;

esp_periph_set_handle_t periph_handle;
esp_periph_set_handle_t periph_set_microphone;
esp_periph_handle_t led_handle;
esp_periph_handle_t wifi_handle;

audio_element_handle_t fatfs_stream_writer;
audio_element_handle_t fatfs_stream_reader;
audio_element_handle_t i2s_stream_writer;
audio_element_handle_t i2s_stream_reader_to_sd;
audio_element_handle_t i2s_stream_reader_to_http;
audio_element_handle_t mp3_decoder;
audio_element_handle_t wav_encoder;
audio_element_handle_t http_stream_writer;`

Also, I think there could be only one **audio_event_iface_set_listener**, because I can't set it for different pipelines in different places

rjam-xav, 2020-11-23:
thank you so much, I tried doing this through separate pipelines but had a problem switching between the two and resource availability of resources. Will look at this again. 
Are you able to keep all your different pipelines running at the same time? or do you need to free up resources? what about starting up the codec/SDCard? the same peripheral can be added to different pipelines and kept running simultaneously?

Saynok, 2020-11-23:
@rjam-xav, I'm using different thread, to control microphone recording 

`
void microphone_record(void *par)
{
    while (1)
    {
        if (needToRecordSeconds > 0)
        {
            ESP_LOGI(MICROPHONE_TAG, "[3.6] Set up  uri (file as fatfs_stream, wav as wav encoder)");
            audio_element_set_uri(fatfs_stream_writer, "/sdcard/rec.wav");

            ESP_LOGI(MICROPHONE_TAG, " Start recording");
            audio_pipeline_run(pipeline_i2s_to_uSD);
            ESP_LOGI(MICROPHONE_TAG, "[ * ] Recording ... %d", needToRecordSeconds);

            vTaskDelay(pdMS_TO_TICKS(1000 * needToRecordSeconds));

            ESP_LOGI(MICROPHONE_TAG, "Stop audio_pipeline");
            audio_pipeline_stop(pipeline_i2s_to_uSD);
            audio_pipeline_wait_for_stop(pipeline_i2s_to_uSD);

            needToRecordSeconds = 0;
        }
        vTaskDelay(pdMS_TO_TICKS(100));
    }
}
` 

also, I'm initialising it in separate file, easier to control what required for each pipeline 

`

void system_init()
{
    esp_log_level_set(TAG_SYSTEM, ESP_LOG_INFO);

    ESP_LOGI(TAG_SYSTEM, "initialisation");

    system_init_periph();
    system_init_led();
    system_init_wifi();
    system_init_fatfs();
    system_init_http();
    system_init_uart();

    system_init_mp3();
    system_init_wav();
    system_init_i2s();

    system_init_i2s_to_sd_pipeline();
    system_init_sd_to_i2s_pipeline();
    system_init_i2s_to_http_pipeline();
}
`

Saynok, 2020-11-23:
Also, try to check log. It creates header for wav file at the stopping of pipeline, so, it should reach this point after some time of recording

ESP_LOGI(TAG, "[ 6 ] Stop audio_pipeline");
audio_pipeline_stop(pipeline);
audio_pipeline_wait_for_stop(pipeline);

rjam-xav, 2020-11-23:
Interesting. I need to be able to keep looping between the wakenet and recording serially. Much like you enter a query for alexa/siri. But I have also realized that the write to file for wav only occurs after the pipeline stops. Is it possible to restart the pipeline without having to relink elements ?

Saynok, 2020-11-23:
Not sure, not using wakenet. For me it's better to keep different pipelines, to avoid any relink.

closed 2020-11-25 by rjam-xav