espressif/esp-adf #513: VOIP-AEC-algorithm_stream issue with IDLE0(CPU 0)

Patsjemoe, 2020-11-24:
Hi,
tried to implement AEC with algorithm stream in the VOIP application, but had below error. Any idea what caused this ? it shows a watchdog trigger, but it is not clear to me where it comes from.

W (16020) SIP: CHANGE STATE FROM 64, TO 72, :func: _sip_uac_process_res:1480
I (16030) SIP: Remote RTP port=28232
I (16030) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_BEGIN
I (16040) VOIP_EXAMPLE: [10.0] Create audio pipeline_rec for recording
I (16050) VOIP_EXAMPLE: [10.1] Create i2s stream to read audio data from codec chip
I (16060) VOIP_EXAMPLE: [10.2] Create algo stream
I (16060) VOIP_EXAMPLE: [10.3] Create sip encoder to encode sip format
I (16070) VOIP_EXAMPLE: [10.4] Create raw stream
I (16070) VOIP_EXAMPLE: [10.5] Register all elements to audio pipeline
I (16080) VOIP_EXAMPLE: [10.6] Link it together
I (16090) VOIP_EXAMPLE:  SIP recorder has been created
I (16090) VOIP_EXAMPLE: [11.0] Create audio pipeline_rec for player
I (16100) VOIP_EXAMPLE: [11.1] Create raw write stream
I (16100) VOIP_EXAMPLE: [11.2] Create sip decoder to decode sip file
I (16110) VOIP_EXAMPLE: [11.3] Create i2s stream to write data to codec chip
W (16120) I2S: I2S driver already installed
I (16120) VOIP_EXAMPLE: [11.4] Register all elements to audio pipeline
I (16130) VOIP_EXAMPLE: [11.5] Link it together
I (16160) VOIP_EXAMPLE: Speaker has been created
E (21020) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21020) task_wdt:  - IDLE0 (CPU 0)
E (21020) task_wdt: Tasks currently running:
E (21020) task_wdt: CPU 0: algo
E (21020) task_wdt: CPU 1: IDLE1
I (24400) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400eaf77  PS      : 0x00060830  A0      : 0x800d45d0  A1      : 0x3ffce970
0x400eaf77: audio_pipeline_stop at C:/path/esp/esp-adf-v2.1/components/audio_pipeline/audio_pipeline.c:89

A2      : 0x00000000  A3      : 0x3f4031e8  A4      : 0x3f403728  A5      : 0x00005f50
A6      : 0x3f4031e8  A7      : 0x3ffe1500  A8      : 0x800d3f58  A9      : 0x3ffce910
A10     : 0x400de788  A11     : 0x3ffb4f30  A12     : 0x3f403728  A13     : 0x0000001f
0x400de788: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/..
.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000001  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

ELF file SHA256: a2111340136009a630a9930355878018359f236d887318738a467ff9da7b007f

Backtrace: 0x400eaf77:0x3ffce970 0x400d45cd:0x3ffce990 0x4016af1b:0x3ffce9d0 0x400f6bf6:0x3ffce9f0 0x4016afab:0x3ffcea
0faa28:0x3ffcea30 0x4009167e:0x3ffcea70
0x400eaf77: audio_pipeline_stop at C:/path/esp/esp-adf-v2.1/components/audio_pipeline/audio_pipeline.c:89

0x400d45cd: _sip_event_handler at C:/path/esp/esp-adf-v2.1/examples/advanced_examples/voip/main/voip_app.

0x4016af1b: _sip_dispach_event at /home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1995

0x400f6bf6: _rtp_callback at /home/michael/work/esp/esp-adf-internal/components/esp_sip/esp_sip.c:1995

0x4016afab: rtp_dispatch_event at /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/sip_rtp.c:130

0x400faa28: _rtp_rx_task at /home/michael/work/esp/esp-adf-internal/components/esp_sip/lib/sip_rtp.c:368

0x4009167e: vPortTaskWrapper at C:/msys32/home/ludok/esp/esp-idf-v3.3.2/components/freertos/port.c:403

BELOW the part I added to VOIP.
I call the pipeline open from the same place as in the voip program when they call recorder and player pipeline.

*********************************************************************
static esp_err_t pipeline_open()
{
    audio_pipeline_handle_t pipeline_rec, pipeline_play;
    esp_log_level_set("*", ESP_LOG_WARN);
    esp_log_level_set(TAG, ESP_LOG_INFO);

    ESP_LOGI(TAG, "[10.0] Create audio pipeline_rec for recording");
    audio_pipeline_cfg_t pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline_rec = audio_pipeline_init(&pipeline_cfg);
    mem_assert(pipeline_rec);

    ESP_LOGI(TAG, "[10.1] Create i2s stream to read audio data from codec chip");
	 
    i2s_stream_cfg_t i2s_rd_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_rd_cfg.type = AUDIO_STREAM_READER;
    audio_element_handle_t i2s_stream_reader = i2s_stream_init(&i2s_rd_cfg);

	ESP_LOGI(TAG, "[10.2] Create algo stream ");
	 
    algorithm_stream_cfg_t algo_config = ALGORITHM_STREAM_CFG_DEFAULT();
    algo_config.input_type = ALGORITHM_STREAM_INPUT_TYPE2;
    audio_element_handle_t element_algo = algo_stream_init(&algo_config);

    ESP_LOGI(TAG, "[10.3] Create sip encoder to encode sip format");
  	g711_encoder_cfg_t g711_enc_cfg = DEFAULT_G711_ENCODER_CONFIG();
    audio_element_handle_t sip_encoder = g711_encoder_init(&g711_enc_cfg);
	
	ESP_LOGI(TAG, "[10.4] Create raw stream ");
	raw_stream_cfg_t raw_rd_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_rd_cfg.type = AUDIO_STREAM_READER;
    raw_read = raw_stream_init(&raw_rd_cfg);
    audio_element_set_output_timeout(raw_read, portMAX_DELAY);

    ESP_LOGI(TAG, "[10.5] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline_rec, i2s_stream_reader, "i2s_rd");
    audio_pipeline_register(pipeline_rec, element_algo, "algo");
	audio_pipeline_register(pipeline_rec, sip_encoder, "sip_encoder");
    audio_pipeline_register(pipeline_rec, raw_read, "raw");

    ESP_LOGI(TAG, "[10.6] Link it together ");
    const char *link_rec[4] = {"i2s_rd", "algo", "sip_encoder", "raw"};
    audio_pipeline_link(pipeline_rec, &link_rec[0], 4);
	audio_pipeline_run(pipeline_rec);
    ESP_LOGI(TAG, " SIP recorder has been created");
	//////////////////////////////////////////////////////////////////////////////////

    ESP_LOGI(TAG, "[11.0] Create audio pipeline_rec for player");
    audio_pipeline_cfg_t pipeline_play_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    pipeline_play = audio_pipeline_init(&pipeline_play_cfg);

    ESP_LOGI(TAG, "[11.1] Create raw write stream ");
    raw_stream_cfg_t raw_wr_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_wr_cfg.type = AUDIO_STREAM_WRITER;
    raw_write = raw_stream_init(&raw_wr_cfg);
    
    ESP_LOGI(TAG, "[11.2] Create sip decoder to decode sip file");
    g711_decoder_cfg_t g711_dec_cfg = DEFAULT_G711_DECODER_CONFIG();
    audio_element_handle_t sip_decoder = g711_decoder_init(&g711_dec_cfg);

    ESP_LOGI(TAG, "[11.3] Create i2s stream to write data to codec chip");
	
    i2s_stream_cfg_t i2s_wd_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_wd_cfg.type = AUDIO_STREAM_WRITER;
    audio_element_handle_t i2s_stream_writer = i2s_stream_init(&i2s_wd_cfg);
	

    ESP_LOGI(TAG, "[11.4] Register all elements to audio pipeline");
    audio_pipeline_register(pipeline_play, raw_write, "raw");
    audio_pipeline_register(pipeline_play, sip_decoder, "sip_decoder");
    audio_pipeline_register(pipeline_play, i2s_stream_writer, "i2s_wd");

    ESP_LOGI(TAG, "[11.5] Link it together ");
    const char *link_tag[3] = {"raw", "sip_decoder", "i2s_wd"};
    audio_pipeline_link(pipeline_play, &link_tag[0], 3);
	audio_pipeline_run(pipeline_play);
	ESP_LOGI(TAG, "Speaker has been created");
	

    //Please reference the way of ALGORITHM_STREAM_INPUT_TYPE2 in "algorithm_stream.h"
    ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
    audio_element_set_multi_output_ringbuf(sip_decoder, input_rb, 0);
	return ESP_OK;
}
thanks in advance
best regards
Ludo

jason-mao, 2020-11-25:
@Patsjemoe 
```c
E (21020) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21020) task_wdt: - IDLE0 (CPU 0)
E (21020) task_wdt: Tasks currently running:
E (21020) task_wdt: CPU 0: algo
E (21020) task_wdt: CPU 1: IDLE1
I (24400) VOIP_EXAMPLE: SIP_EVENT_AUDIO_SESSION_END
Guru Meditation Error: Core 0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
```

This reset caused by algorithm task not feed watchdog.  Look like cpu0 is too busy? `audio_sys_get_real_time_stats` can get more informations of task.

Patsjemoe, 2020-11-25:
hi,
thank you for yor quick reply.
Unfortunately, I am not familiar with audio_sys_get_real_time_stats, and could not find it in the docs.
where can I find more information on how to use this ? (in the VOIP app)
If the core is already overloaded, will this then work ?
thanks in advance
regards
Ludo

jason-mao, 2020-11-26:
@Patsjemoe The audio_sys_get_real_time_stats  function implement as $ADF_PATHcomponents/audio_sal/audio_sys.c, useage refer to :https://github.com/espressif/esp-adf/blob/3214230c8234e039cfc5d42cbd8dedc6f62fb5e9/examples/cli/main/console_example.c#L415.  You should move some task to others core to reduce the overloaded core.

Patsjemoe, 2020-11-26:
@jason-mao 
thank you.

I moved algorithm stream + I2s stream to core 1, and still have core 0 panic'd.  
I keep looking. 
************************************************************************************************************

algorithm stream to core1 
-------------------------
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400eb017  PS      : 0x00060c30  A0      : 0x800d4670  A1      : 0x3ffcea40
0x400eb017: audio_pipeline_stop at C:/msys32/home/ludok/esp/esp-adf-v2.1/components/audio_pipeline/audio_pipeline.
c:893

A2      : 0x00000000  A3      : 0x3f4031e8  A4      : 0x3f40375c  A5      : 0x00005d35
A6      : 0x3f4031e8  A7      : 0x3ffec060  A8      : 0x800d3fa4  A9      : 0x3ffce9e0
A10     : 0x400de828  A11     : 0x3ffb4f30  A12     : 0x3f40375c  A13     : 0x0000001f
0x400de828: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdi
o/../../../.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000001  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

ELF file SHA256: d0cba6dbc1bce7b95de74e0061d9f421616ae9b85266100178ddab9afd3840ae

Backtrace: 0x400eb017:0x3ffcea40 0x400d466d:0x3ffcea60 0x4016b3ab:0x3ffceaa0 0x400f7062:0x3ffceac0 0x4016b43b:0x3f
fceae0 0x400faeb0:0x3ffceb00 0x4009167e:0x3ffceb40

+ I2S stream to core 1 
----------------------
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400eb017  PS      : 0x00060c30  A0      : 0x800d4670  A1      : 0x3ffce970
0x400eb017: audio_pipeline_stop at C:/msys32/home/ludok/esp/esp-adf-v2.1/components/audio_pipeline/audio_pipeline.
c:893

A2      : 0x00000000  A3      : 0x3f4031e8  A4      : 0x3f40375c  A5      : 0x00004665
A6      : 0x3f4031e8  A7      : 0x00000000  A8      : 0x800d3fa4  A9      : 0x3ffce910
A10     : 0x400de828  A11     : 0x3ffb4f30  A12     : 0x3f40375c  A13     : 0x0000001f
0x400de828: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdi
o/../../../.././newlib/libc/stdio/vprintf.c:35

A14     : 0x00000001  A15     : 0x00000003  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000010  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc

ELF file SHA256: 590d0343094842dfa34acc54520cbf5b79761c396025ca4dfaac39284d6e9b42

Backtrace: 0x400eb017:0x3ffce970 0x400d466d:0x3ffce990 0x4016b3ab:0x3ffce9d0 0x400f7062:0x3ffce9f0 0x4016b43b:0x3f
fcea10 0x400faeb0:0x3ffcea30 0x4009167e:0x3ffcea70

Note : 
I found : 
static esp_err_t task_real_time_states(esp_periph_handle_t periph, int argc, char *argv[])
{
    audio_sys_get_real_time_stats();
    return ESP_OK;
}
can you advise what to use for :  esp_periph_handle_t periph, int argc, char *argv[] ?
regards
Ludo

Patsjemoe, 2020-11-27:
@jason-mao 
I used adf-v2.2, added the algorithm_stream to it and tested VOIP, without core 0 panic. In this test, I have put algo in core 1 (#define ALGORITHM_STREAM_PINNED_TO_CORE   1 in algorithm_stream.h ), and deleted I2C display tasks and  power modem sleep,
I can hear my voice from GSM to esp32, but not yet from esp32 to phone. 
 I will need to do now further testing on adding these back into the application. 
Do you have any idea on how much % modem sleep is taking from the core ?
I would like to get some statistics with  task_real_time_states, but am stuck on how to use it in the VOIP application. Any suggestion would be welcome.
Thanks already for your help.
regards
Ludo

closed 2020-12-02 by Patsjemoe