espressif/esp-adf #514: connect to access point without using menuconfig

rjam-xav, 2020-11-25:
Hello,

Is there a way to connect to an access point without using menuconfig? I tried using the guide [here](https://github.com/espressif/esp-idf/tree/722043f734fa556d66d57473ac266fb1d0ec5ad2/examples/wifi/getting_started/station) but because the adf imports only certain liibraries of the esp-idf some dependencies are not satisfied. 
I specifically get the error:

>  ./main/srcs/wifi_connect.c: In function 'event_handler':
> ../main/srcs/wifi_connect.c:53:25: warning: comparison between pointer and integer
>          if (s_retry_num < EXAMPLE_ESP_MAXIMUM_RETRY) {
>                          ^
> ../main/srcs/wifi_connect.c:53:25: error: comparison with string literal results in unspecified behavior [-Werror=address]
> In file included from C:/esp/esp-idf/components/esp_common/include/esp_timer.h:44,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portmacro.h:84,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portable.h:96,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
>                  from ../main/srcs/wifi_connect.c:11:
> ../main/srcs/wifi_connect.c: In function 'wifi_init_sta':
> ../main/srcs/wifi_connect.c:75:21: error: implicit declaration of function 'esp_netif_init'; did you mean 'esp_wifi_init'? [-Werror=implicit-function-declaration]
>      ESP_ERROR_CHECK(esp_netif_init());
>                      ^~~~~~~~~~~~~~
> C:/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
>          esp_err_t __err_rc = (x);                                       \
>                                ^
> ../main/srcs/wifi_connect.c:78:5: error: implicit declaration of function 'esp_netif_create_default_wifi_sta' [-Werror=implicit-function-declaration]
>      esp_netif_create_default_wifi_sta();
>      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> ../main/srcs/wifi_connect.c:83:5: error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?
>      esp_event_handler_instance_t instance_any_id;
>      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
>      esp_event_handler_t
> ../main/srcs/wifi_connect.c:84:5: error: unknown type name 'esp_event_handler_instance_t'; did you mean 'esp_event_handler_t'?
>      esp_event_handler_instance_t instance_got_ip;
>      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
>      esp_event_handler_t
> In file included from C:/esp/esp-idf/components/esp_common/include/esp_timer.h:44,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portmacro.h:84,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portable.h:96,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
>                  from ../main/srcs/wifi_connect.c:11:
> ../main/srcs/wifi_connect.c:85:21: error: implicit declaration of function 'esp_event_handler_instance_register'; did you mean 'esp_event_handler_unregister'? [-Werror=implicit-function-declaration]
>      ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
>                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> C:/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
>          esp_err_t __err_rc = (x);                                       \
>                                ^
> ../main/srcs/wifi_connect.c:105:14: error: 'wifi_sta_config_t' {aka 'struct <anonymous>'} has no member named 'pmf_cfg'
>              .pmf_cfg = {
>               ^~~~~~~
> ../main/srcs/wifi_connect.c:105:24: error: extra brace group at end of initializer
>              .pmf_cfg = {
>                         ^
> ../main/srcs/wifi_connect.c:105:24: note: (near initialization for 'wifi_config.sta')
> ../main/srcs/wifi_connect.c:105:24: warning: excess elements in struct initializer
> ../main/srcs/wifi_connect.c:105:24: note: (near initialization for 'wifi_config.sta')
> In file included from C:/esp/esp-idf/components/esp_common/include/esp_timer.h:44,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portmacro.h:84,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/portable.h:96,
>                  from C:/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
>                  from ../main/srcs/wifi_connect.c:11:
> ../main/srcs/wifi_connect.c:140:21: error: implicit declaration of function 'esp_event_handler_instance_unregister'; did you mean 'esp_event_handler_unregister'? [-Werror=implicit-function-declaration]
>      ESP_ERROR_CHECK(esp_event_handler_instance_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, instance_got_ip));
>                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> C:/esp/esp-idf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
>          esp_err_t __err_rc = (x);                                       \
>                                ^
> cc1.exe: some warnings being treated as errors
> ninja: build stopped: subcommand failed.
> ninja failed with exit code 1
>

jason-mao, 2020-11-26:
@rjam-xav Which IDF version did you used?  Looks like due to the warnings `cc1.exe: some warnings being treated as errors`.

rjam-xav, 2020-11-26:
I was able to get it running with this:
static const char* APTAG = "AP_Connect";
#define EXAMPLE_ESP_WIFI_SSID      "8629"
#define EXAMPLE_ESP_WIFI_PASS      "v-403S87"

> void connect_main(void)
> {
>     esp_err_t err = nvs_flash_init();
>     if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
>         // NVS partition was truncated and needs to be erased
>         // Retry nvs_flash_init
>         ESP_ERROR_CHECK(nvs_flash_erase());
>         err = nvs_flash_init();
>     }
>     tcpip_adapter_init();
> 
>     ESP_LOGI(APTAG, "[ 1 ] Start and wait for Wi-Fi network");
>     esp_periph_config_t w_periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
>     esp_periph_set_handle_t w_set = esp_periph_set_init(&w_periph_cfg);
>     periph_wifi_cfg_t wifi_cfg = {
>         .ssid = EXAMPLE_ESP_WIFI_SSID,
>         .password = EXAMPLE_ESP_WIFI_PASS,
>     };
>     esp_periph_handle_t wifi_handle = periph_wifi_init(&wifi_cfg);
>     esp_periph_start(w_set, wifi_handle);
>     periph_wifi_wait_for_connected(wifi_handle, portMAX_DELAY);
> }

closed 2020-11-26 by rjam-xav