espressif/esp-adf #516: Bad audio data from I2S bus

nixmeer, 2020-11-25:
Hello :)

I am trying to write audio data to a buffer using a LyraT v4.3. The code is the following (from TFL micro speech):
```cpp
  i2s_config_t i2s_config = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_TX),
      .sample_rate = 16000,
      .bits_per_sample = (i2s_bits_per_sample_t)16,
      .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,
      .communication_format = I2S_COMM_FORMAT_I2S,
      .intr_alloc_flags = 0,
      .dma_buf_count = 3,
      .dma_buf_len = 300,
      .use_apll = false,
      .tx_desc_auto_clear = false,
      .fixed_mclk = -1,
  };
  i2s_pin_config_t pin_config = {
      .bck_io_num = 5,    // IIS_SCLK
      .ws_io_num = 25,     // IIS_LCLK
      .data_out_num = -1,  // IIS_DSIN
      .data_in_num = 35,   // IIS_DOUT
  };
  esp_err_t ret = 0;
  ret = i2s_driver_install((i2s_port_t)1, &i2s_config, 0, NULL);
  if (ret != ESP_OK) {
    ESP_LOGE(TAG, "Error in i2s_driver_install");
  }
  ret = i2s_set_pin((i2s_port_t)1, &pin_config);
  if (ret != ESP_OK) {
    ESP_LOGE(TAG, "Error in i2s_set_pin");
  }

  ret = i2s_zero_dma_buffer((i2s_port_t)1);
  if (ret != ESP_OK) {
    ESP_LOGE(TAG, "Error in initializing dma buffer with 0");
  }
```
This is called in a loop:
```cpp
  size_t bytes_read;
  uint8_t i2s_read_buffer[i2s_bytes_to_read] = {};
  i2s_read((i2s_port_t)1, (void *)i2s_read_buffer, i2s_bytes_to_read,
             &bytes_read, 10);
```
Pin numbers are chosen based on the [schematic](https://dl.espressif.com/dl/schematics/esp32-lyrat-v4.3-schematic.pdf). 

Many received data is just `0x00`, the rest does not seem plausible. 

How do I test the audio stream? Can I somehow re-direct the data from the buffer (without changing it to a stream) back to the audio chip as output?

jason-mao, 2020-12-02:
@nixmeer Hi, Could you please provide the data path? I have  not understand logical relationship, ` write audio data to a buffer using a LyraT v4.3.` and why call  `i2s_read((i2s_port_t)1, (void *)i2s_read_buffer, i2s_bytes_to_read, &bytes_read, 10); `.

nixmeer, 2020-12-02:
@jason-mao I try to read the data from one of the microphones. Not only does I2S need to be configured, but also the audio codec chip. 
```cpp
static audio_hal_codec_i2s_iface_t i2s_iface = {
  .mode = AUDIO_HAL_MODE_MASTER,        /*!< audio codec chip mode */
  .fmt = AUDIO_HAL_I2S_LEFT,       /*!< I2S interface format */
  .samples = AUDIO_HAL_16K_SAMPLES,  /*!< I2S interface samples per second */
  .bits = AUDIO_HAL_BIT_LENGTH_16BITS,        /*!< i2s interface number of bits per sample */
};
static audio_hal_codec_config_t es8388_config = {
  .adc_input = AUDIO_HAL_ADC_INPUT_LINE1,
  .dac_output = AUDIO_HAL_DAC_OUTPUT_ALL,
  .codec_mode = AUDIO_HAL_CODEC_MODE_ENCODE,
  .i2s_iface = i2s_iface,
};
static const es_module_t es8388_mode = ES_MODULE_ADC;

es8388_init(&es8388_config);
es8388_start(es8388_mode);
```
Somehow, some examples avoid this (maybe by using a board_initialize function). In my opinion, the documentation is not very useful on that. There are some types that are not referenced (`i2s_config_t` [here](https://docs.espressif.com/projects/esp-adf/en/latest/api-reference/streams/#structures) i.e.)

Some defaults seem to have changed over time (`AUDIO_HAL_ES8388_DEFAULT()` i.e. can be found in some code but I could not find no definition).

However, after I solved this issue, I used a pipeline with an `i2s_stream` and a `raw_stream` and the data looks valid. But since I could not find a way to figure out how much data is available to be read from the stream, I had to create another task reading the `raw_steam` and saving it to another ringbuffer while adding up how much data was copied...

nixmeer, 2020-12-22:
ok, I managed to get it running. The tensorflow example I tried to run did only initialize the i2s bus on the ESP but did not initialize the ES8388. The following works:

```cpp
audio_hal_codec_config_t audio_codec_cfg = {
  .adc_input = AUDIO_HAL_ADC_INPUT_LINE1,
  .dac_output = AUDIO_HAL_DAC_OUTPUT_ALL,
  .codec_mode = AUDIO_HAL_CODEC_MODE_BOTH,
  .i2s_iface = {
      .mode = AUDIO_HAL_MODE_SLAVE,
      .fmt = AUDIO_HAL_I2S_NORMAL,
      .samples = AUDIO_HAL_16K_SAMPLES,
      .bits = AUDIO_HAL_BIT_LENGTH_16BITS,
  },  
};

static audio_element_handle_t i2s_stream_reader;
static i2s_stream_cfg_t i2s_cfg = {
    .type = AUDIO_STREAM_READER,                                                
    .i2s_config = {                                                             
        .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_RX),                    
        .sample_rate = (int16_t)16000,                                                   
        .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,                           
        .channel_format = I2S_CHANNEL_FMT_ONLY_LEFT,                           
        .communication_format = I2S_COMM_FORMAT_I2S,                            
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL2 | ESP_INTR_FLAG_IRAM,          
        .dma_buf_count = 3,                                                     
        .dma_buf_len = 320,                                                     
        .use_apll = true,                                                       
        .tx_desc_auto_clear = true,                                             
        .fixed_mclk = 0                                                         
    },                                                                          
    .i2s_port = I2S_NUM_0,                                                      
    .use_alc = false,                                                           
    .volume = 0,
    .out_rb_size = I2S_STREAM_RINGBUFFER_SIZE,                                  
    .task_stack = I2S_STREAM_TASK_STACK,                                        
    .task_core = I2S_STREAM_TASK_CORE,                                          
    .task_prio = I2S_STREAM_TASK_PRIO,                                          
    .stack_in_ext = false,                                                      
    .multi_out_num = 0,                                                         
    .uninstall_drv = true,                                                      
};

static audio_element_handle_t raw_read;
static raw_stream_cfg_t raw_cfg = {
      .type = AUDIO_STREAM_READER,
      .out_rb_size = 8 * 1024,
};

void InitAudioRecording() {
  audio_hal_handle_t codec_hal = audio_hal_init(&audio_codec_cfg, &AUDIO_CODEC_ES8388_DEFAULT_HANDLE);
  audio_hal_ctrl_codec(codec_hal, AUDIO_HAL_CODEC_MODE_BOTH, AUDIO_HAL_CTRL_START);

  pipeline_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
  pipeline = audio_pipeline_init(&pipeline_cfg);

  i2s_stream_reader = i2s_stream_init(&i2s_cfg);

  raw_read = raw_stream_init(&raw_cfg);

  audio_pipeline_register(pipeline, i2s_stream_reader, "i2s");
  audio_pipeline_register(pipeline, raw_read, "raw");
  audio_pipeline_link(pipeline, &link_tag[0], 2);

  audio_event_iface_cfg_t evt_cfg = AUDIO_EVENT_IFACE_DEFAULT_CFG();
  audio_event_iface_handle_t evt = audio_event_iface_init(&evt_cfg);
  audio_pipeline_set_listener(pipeline, evt);

  audio_pipeline_run(pipeline);
}
```

After calling `InitAudioRecording`, `raw_stream_read(raw_read, raw_buffer, sizeof(raw_buffer))` works as expected.

closed 2020-12-22 by nixmeer