espressif/esp-adf #517: reading a wav file using fopen.

rjam-xav, 2020-11-26:
Hello,

I am reading a wav file in bytes as shown in the example [here](https://github.com/espressif/esp-adf/blob/3214230c8234e039cfc5d42cbd8dedc6f62fb5e9/examples/player/element_sdcard_mp3/main/element_sdcard_mp3_example.c)

However, I always get a could not open file error. What can be the reason for this?

My code:

`   static FILE* file;
    if (file == NULL) {
        file = fopen("/sdcard/REC.wav", "rb");
        if (!file) {
            printf("Error opening file\n");
            return 0;
        }
    }
    fseek(file, 0L, SEEK_END);
    long int res = ftell(file);

    fseek(file, 0, SEEK_SET);
    char buf[res];

    int read_len = fread(buf, res,1, file);
`

jason-mao, 2020-11-26:
@rjam-xav  I think due to the long name of FATFS. Please set the `sdkconfig` as below and test.

```c
CONFIG_FATFS_CODEPAGE_936=y
CONFIG_FATFS_CODEPAGE=936
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FATFS_MAX_LFN=255
CONFIG_FATFS_API_ENCODING_UTF_8=y
CONFIG_FATFS_FS_LOCK=0
CONFIG_FATFS_TIMEOUT_MS=10000
CONFIG_FATFS_PER_FILE_CACHE=y
```

rjam-xav, 2020-11-26:
Thank you. I tried doing that but this is not changing. My Sdcard is mounted. below is the code and the log:

>    ESP_LOGI(PTAG, "[ 1 ] Mount sdcard");
>     // Initialize peripherals management
>     esp_periph_config_t periph_cfg = DEFAULT_ESP_PERIPH_SET_CONFIG();
>     esp_periph_set_handle_t set = esp_periph_set_init(&periph_cfg);
> 
>     // Initialize SD Card peripheral
>     audio_board_sdcard_init(set, SD_MODE_1_LINE);
> 
>     ESP_LOGW(PTAG,"mounted");
> 
>     static FILE* file;
>     if (file == NULL) {
>         file = fopen("/sdcard/REC.wav", "rb");
>         if (!file) {
>             printf("Error opening file\n");
>             return 0;
>         }
>         else
>             printf("File Open Success");
>     }
> 
>     fseek(file, 0L, SEEK_END);
>     long int res = ftell(file);
> 
>     fseek(file, 0, SEEK_SET);
>     char buf[res];
> 
>     int read_len = fread(buf, res, 1, file);
>     if (read_len == 0) {
>         fclose(file);
>         file = NULL;
>         //read_len = 0;
>     }
>     ESP_LOGI(PTAG, "[ * ] bytes in msg, data_len=%d, ", read_len);

and the error:

> 
> rst:0x8 (TG1WDT_SYS_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:7284
> load:0x40078000,len:14944
> load:0x40080400,len:4988
> entry 0x40080708
> I (323) boot: Chip Revision: 1
> I (391) boot_comm: chip revision: 1, min. bootloader chip revision: 0
> W (237) boot: PRO CPU has been reset by WDT.
> W (238) boot: WDT reset info: PRO CPU PC=0x40097c99
> 0x40097c99: _xt_context_save at C:/esp/esp-idf/components/freertos/xtensa_context.S:168
> 
> W (238) boot: WDT reset info: APP CPU PC=0x4009210f
> 0x4009210f: panicHandler at C:/esp/esp-idf/components/esp32/panic.c:245 (discriminator 1)

rjam-xav, 2020-11-26:
Update:
When I change the fopen command to file = fopen("file://sdcard/dingdong.mp3", "rb"); it returns error opening file 

>  I (22424) HTTP_CLIENT: [ 1 ] Mount sdcard
> W (22929) HTTP_CLIENT: mounted
> Error opening file

rjam-xav, 2020-11-26:
Just as an alternate is there also another way to read the wav file without using the fatfs stream writer element and constructing a pipeline?

rjam-xav, 2020-11-26:
Hello, I just realized that every time I build my project the sdkconfig file gets overwritten perhaps by the default settings in menuconfig.

rjam-xav, 2020-11-26:
Update: configured the sdkconfig as suggested but still getting error opening file

jason-mao, 2020-12-02:
@rjam-xav How is going? Did you can be opened other files?

rjam-xav, 2020-12-02:
@jason-mao , yes it is working now, I scratched everything and wrote code from scratch. most likely there was another issue that I have still not been able to identify. but thank you for your help.

closed 2020-12-02 by rjam-xav