espressif/esp-adf #519: navigation in audio track

mcSimoffa, 2020-11-27:
How I can move from position 'byte X' to position 'byte Y' in *.m4a media file?
I use audioelement aac decoder. 
audio_element_set_byte_pos() function not work. So it accept my new position  without error,  but the music don't jump to the new position
Do you have any example about this ?

jason-mao, 2020-12-02:
@mcSimoffa The m4a seek is a complex operation. We provide  high level APIs in `esp_audio.h`, usage refer to https://github.com/espressif/esp-adf/blob/73a2ec52436ce30195c010c05c9382bfac09eadc/examples/cli/main/console_example.c#L187

closed 2020-12-02 by jason-mao

mcSimoffa, 2020-12-04:
Jason, thanks for you help, but I need a little bigger.
I need to navigate with accurate about 22ms (one m4a block = 1024 samle. sample rate = 44100). esp_audio_seek() have an  accurate about 1 sec. So I develope such alogritm:
//here music is sound
audio_element_pause(aac_decoder);
rb_reset(ringbuf);
m4a_seek(position, &bytePos, );  /* this is my function. It parse atoms 'stts' and 'stsz' and calculate the position in 'mdat' atom. This position I use in callback: read_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx) function for fseek() function. */
audio_element_set_byte_pos(aac_decoder, bytePos);
audio_element_resume(aac_decoder,0,100); //decoder wake up and read the block which need for me!
Such I feed the decoder a need block. It work, the music jump! It have positioning with absolutly accurate. 
But there is two bad point:
1. I cath a error 'M4A_PARSER: can not seach the sample ID , line 1048'  after jump. it doesn't matter because it continue work, the musiccan play and jump many times.
2. But when time passes equal to the duration of my song, then the decoder will stop. It can stop in the middle of the song sounded (I mooved the song back several times). It stops. Obviously, I must reset the counter of decoded samples. 
But how can I do this?
Even if I use a standard seek function 'audio_element_seek()' - my music also stopped before reaching the end of the track. With message AAC_DECODER: Closed by [2]. I want to stop music only at end of track, when callback: read_cb() back EOF.
msg source type 131072   msg source 0x3f810e04   msg cmd 8   msg data 0xf
0x3f810e04  - it's AAC decoder handle
The aac decoder source file have *.a shape. I don't see code, I don't know how it work in detail. If it's easy say me please: how to assign a sample counter to new value