espressif/esp-adf #521: esp-adf patched for esp-idf version release/4.2

huybk213, 2020-11-30:
Dear everyone,

I want to use esp32s2 board for audio project, but current adf release (version 2.2) does not support path for esp-adf 4.2.
It's only support idf_v3.3_freertos.patch.

Could you please share your patch for esp-idf version 4.2?

Thank you

jason-mao, 2020-12-02:
@huybk213  The  idf_v3.3_freertos.patch can be apply on v4.2. Sorry for confuse with the name of patch.

closed 2020-12-02 by jason-mao

huybk213, 2020-12-02:
Thanks @jason-mao

huybk213, 2020-12-02:
Hi @jason-mao 
After applied idf-patch, I couldn't compile project success.

Because compiler didn't see function xTaskCreateRestrictedPinnedToCore.
After take a look at your  idf_v3.3_freertos.patch and file "task.h", "task.c", it seem macro portUSING_MPU_WRAPPERS not set before, so function xTaskCreateRestrictedPinnedToCore is not enabled.
Note that in project sdkconfig file, the variable **CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY=y** and **CONFIG_SPIRAM_BOOT_INIT=y**

How can resolve the problem?

Thank you.

jason-mao, 2020-12-02:
@huybk213 How is the error? which example you used?

huybk213, 2020-12-02:
Dearn @jason-mao
I use hello-world example in esp-idf
Because in esp-adf components, audio_thread.c contain a weak function xTaskCreateRistrictedPintoCore.
So i want to test FreeRTOS patch by hello world example.

My code is here

'#include <stdio.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

void test_task_in_ext(void * param)
{
    while (1)
    {
        printf("Hello world for ext ram\r\n");
        xTaskDelay(1000);
    }
    
}

void app_main(void)
{
    printf("Hello world!\n");

    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU cores, WiFi%s%s, ",
            CONFIG_IDF_TARGET,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Free heap: %d\n", esp_get_free_heap_size());

    fflush(stdout);

    StackType_t task_stack;
    memset(&task_stack, 0, sizeof(StackType_t));
    /*
        * Note: 1. ESP32-ECO3 chip support stack on external memory only.
        *       2. Make sure selected the `CONFIG_SPIRAM_BOOT_INIT` and `CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY` by `make menuconfig`
        *       3. Please apply the $ADF_PATH/idf_patches/idf_v3.3_freertos.patch
        *
        */
    ESP_LOGI(TAG, "The %s task allocate stack on external memory", name);
    TaskParameters_t xRegParameters = {
        .pvTaskCode = main_func,
        .pcName = name,
        .usStackDepth = 8192,
        .pvParameters = NULL,
        .uxPriority = 6 | portPRIVILEGE_BIT,
        .puxStackBuffer = task_stack,
        .xRegions = {{
                .pvBaseAddress = 0x00,
                .ulLengthInBytes = 0x00,
                .ulParameters = 0x00,
            }
        }
    };
    if (xTaskCreateRestrictedPinnedToCore(&xRegParameters, (xTaskHandle)&p_handle, core_id) != pdPASS) {
[... 1 lines cut ...]
        ESP_LOGE(TAG, "Error creating RestrictedPinnedToCore %s", name);
        esp_restart();
    }
    while(1)
    {
        vTaskDelay(1000);
    }

    // esp_restart();
}
`

Console log here

`[775/780] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
/home/huytv/.espressif/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32s2/. -I../../../../components/soc/src/esp32s2/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32s2/include -I../../../../components/esp32s2/include -I../../../../components/driver/include -I../../../../components/driver/esp32s2/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32s2/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32s2/include -I../../../../components/soc/soc/esp32s2/../include -I../../../../components/soc/soc/esp32s2/private_include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32s2/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/in [... cut ...]
../main/hello_world_main.c: In function 'app_main':
../main/hello_world_main.c:80:9: error: implicit declaration of function 'xTaskCreateRestrictedPinnedToCore'; did you mean 'xTaskCreateStaticPinnedToCore'? [-Werror=implicit-function-declaration]
     if (xTaskCreateRestrictedPinnedToCore(&xRegParameters, &handle, 0) != pdPASS) {
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         xTaskCreateStaticPinnedToCore
cc1: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

huybk213, 2020-12-07:
Hi @jason-mao 

Could you please give me your suggestion?

Thanks

jason-mao, 2020-12-07:
@huybk213 Did you used the esp32s2 chip?

huybk213, 2020-12-07:
> @huybk213 Did you used the esp32s2 chip?

Yes, i used esp32s2, board esp32 kaluga-v1.3

jason-mao, 2020-12-07:
I  think due to  portUSING_MPU_WRAPPERS is unavailable on S2. Let me check.

huybk213, 2020-12-14:
hi @jason-mao 
Do you have any update?

jason-mao, 2020-12-15:
@huybk213 You can set the portUSING_MPU_WRAPPERS true manually. We will found a fix way to update the patches.

huybk213, 2020-12-15:
Thanks @jason-mao