espressif/esp-adf #522: Could I make headphone with ESP32 LyraT kit have audio,MIC and receive notification with ANCS? (IDFGH-4345)

quangdoict, 2020-12-01:
Hello,
I try to used esp32 LyraT kit with example ble_ancs and got notification. But now, I want to when accept ImcomingCall I can use esp32 LyraT as headphone as audio, MIC.
Please help me share your opinion.
Thank so much!

HengYongChao, 2020-12-15:
@quangdoict  This is certainly possible, and your requirements actually include several features in ADF. such as BT-a2DP, BLE-ancs. The two main examples are `https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_a2dp_sink_stream` and `https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_ancs` .  You need to combine them as a project.

quangdoict, 2020-12-16:
Thank for your reply!
I try combination adf and ancs but it so complex. if can, please guide me the way combine it.
thank so much!

HengYongChao, 2020-12-17:
@quangdoict 
A good approach is to familiarize yourself with the two individual examples, understand their general code structure and dependencies, and then try to integrate them.

closed 2021-09-26 by jason-mao