espressif/esp-adf #523: idf.py menuconfig: cmake: No such file or directory

KISSMonX, 2020-12-01:
```zsh
➜  ESP32_LyraT_Player git:(master) idf.py menuconfig

Executing action: menuconfig
Running ninja in directory /Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build
Executing "ninja menuconfig"...
[0/1] Re-running CMake...
/bin/sh: /usr/local/Cellar/cmake/3.19.0/bin/cmake: No such file or directory
FAILED: build.ninja
/usr/local/Cellar/cmake/3.19.0/bin/cmake --regenerate-during-build -S/Users/mingh/eclipse-workspace/ESP32_LyraT_Player -B/Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
```

KISSMonX, 2020-12-01:
rm -rf build 

then these error:

```bash
➜  ESP32_LyraT_Player git:(master) ✗ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /Users/mingh/eclipse-workspace/ESP32_LyraT_Player"...
-- Found Git: /usr/local/bin/git (found version "2.29.2")
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/mingh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/mingh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /Users/mingh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- broken
CMake Error at /Users/mingh/.espressif/tools/cmake/3.16.4/CMake.app/Contents/share/cmake-3.16/Modules/CMakeTestCCompiler.cmake:60 (message):
  The C compiler

    "/Users/mingh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build/CMakeFiles/CMakeTmp

    Run Build Command(s):/Users/mingh/.espressif/tools/ninja/1.10.0/ninja cmTC_c029a && [1/2] Building C object CMakeFiles/cmTC_c029a.dir/testCCompiler.c.obj
    [2/2] Linking C executable cmTC_c029a
    FAILED: cmTC_c029a
    : && /Users/mingh/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -mlongcalls -Wno-frame-address  -L/usr/local/opt/qt/lib CMakeFiles/cmTC_c029a.dir/testCCompiler.c.obj  -o cmTC_c029a   && :
    /Users/mingh/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
    /Users/mingh/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
    /Users/mingh/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
    /Users/mingh/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
    /Users/mingh/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
    collect2: error: ld returned 1 exit status
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /Users/mingh/esp/esp-idf/tools/cmake/project.cmake:308 (__project)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build/CMakeFiles/CMakeOutput.log".
See also "/Users/mingh/eclipse-workspace/ESP32_LyraT_Player/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

jason-mao, 2020-12-02:
@KISSMonX Which example did you build?

KISSMonX, 2020-12-02:
> @KISSMonX Which example did you build? And IDF v4.3 should be use the gcc-8-2-0.

examples/player/pipeline_sdcard_mp3_control

KISSMonX, 2020-12-02:
> @KISSMonX Which example did you build? And IDF v4.3 should be use the gcc-8-2-0 at least.

ok，i will try again

KISSMonX, 2020-12-02:
switch to Windows dev environment.  
closed

closed 2020-12-02 by KISSMonX