espressif/esp-adf #524: ESP32_LyraT  vol- canot work?

KISSMonX, 2020-12-01:
ESP32-LyraT V4.3

**SD Card Mode set 1_Line**

```c
audio_board_sdcard_init(set, SD_MODE_1_LINE);
```

[image]

[image]

KISSMonX, 2020-12-01:
[image]

HengYongChao, 2022-02-09:
Hi @KISSMonX 

According to [the schematic diagram of ESP32-Lyrat v4.3](https://docs.espressif.com/projects/esp-adf/en/latest/design-guide/dev-boards/board-esp32-lyrat-v4.3.html), if you need to use the touch button vol- , then you should set the multiplexed key GPIO13 back to the OFF position.
[image]

And I've tested, the touch function of VOL- can be used normally.
[image]

Please try as I mentioned.

closed 2023-09-05 by jason-mao