espressif/esp-adf #525: AUDIO_THREAD: Error creating task input_key_service

jhancharler, 2020-12-01:
I'm trying to run a SIP VOIP demo.

My code compiles and flashes, but it crashes whilst running.

This is the log that I see:

Please note I am using Olimex's ESP-ADF board and SIP VOIP demo software. I'm posting here because the issue seems to be from the ESP-ADF library.

Many thanks in advance!

```
I (2319) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5096) event: sta ip: 192.168.43.194, mask: 255.255.255.0, gw: 192.168.43.1
I (5097) SIP_PHONE: [ 2 ] Start codec chip
I (5119) ES8388_DRIVER: init,out:02, in:00
I (5126) AUDIO_HAL: Codec mode is 3, Ctrl:1
I (5130) SIP_PHONE: [ 3 ] Create and start input key service
E (5130) AUDIO_THREAD: Error creating task input_key_service
E (5133) PERIPH_SERVICE: Create task failed on periph_service_create
E (5140) INPUT_KEY_SERVICE: /home/diogo/Documents/Jhan/esp-adf/components/input_key_service/input_key_service.c:214 (input_key_service_create): Got NULL Pointer
E (5155) INPUT_KEY_SERVICE: /home/diogo/Documents/Jhan/esp-adf/components/input_key_service/input_key_service.c:158 (input_key_service_add_key): Got NULL Pointer
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4019b28b  PS      : 0x00060830  A0      : 0x800d7b4c  A1      : 0x3ffcd920
0x4019b28b: periph_service_set_callback at /home/diogo/Documents/Jhan/esp-adf/components/esp_dispatcher/periph_service.c:125

A2      : 0x00000000  A3      : 0x400d7668  A4      : 0x3f800fd0  A5      : 0x3ffcd9d4
0x400d7668: input_key_service_cb at /home/diogo/Documents/Jhan/sip_phone_example/main/voip_app.c:389

A6      : 0x00000005  A7      : 0x00000000  A8      : 0x801393f4  A9      : 0x3ffcd8c0
A10     : 0x00000001  A11     : 0x3f424cbc  A12     : 0x3f40de68  A13     : 0x00001423
A14     : 0x3f424cbc  A15     : 0x3f424c68  SAR     : 0x00000004  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000010  LBEG    : 0x4008fc61  LEND    : 0x4008fc71  LCOUNT  : 0xfffffffb
0x4008fc61: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84

0x4008fc71: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96

```

jason-mao, 2020-12-02:
@jhancharler Is there have param on Olimex's ESP-ADF board ?  Most of reason for `E (5130) AUDIO_THREAD: Error creating task input_key_service` is due to no memory.

jhancharler, 2020-12-02:
@jason-mao Hello, thank you for your response. What kind of param? 

I will show you menuconfig settings, please see picture:

[image]

jason-mao, 2020-12-02:
@jhancharler Sorry for typo.It's PSRAM, not param. And  please add `AUDIO_MEM_SHOW()` before  error message.