espressif/esp-adf #526: max volume cause panic reset?

KISSMonX, 2020-12-01:
Exceeds USB power supply capacity？？

### log screen: 

[image]

### run log: 

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:13072
load:0x40080400,len:3896
entry 0x40080688
[0;32mI (31) boot: ESP-IDF v4.1-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 23:56:28[0m
[0;32mI (31) boot: chip revision: 1[0m
[0;32mI (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (43) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (46) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (55) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (64) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (79) boot:  1 phy_init         RF data          01 01 0000d000 00001000[0m
[0;32mI (86) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (94) boot: End of partition table[0m
[0;32mI (98) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x3f414 (259092) map[0m
[0;32mI (212) esp_image: segment 1: paddr=0x0004f43c vaddr=0x3ffbdb60 size=0x00bdc (  3036) load[0m
[0;32mI (214) esp_image: segment 2: paddr=0x00050020 vaddr=0x400d0020 size=0x3c308 (246536) map[0m
[0;32mI (313) esp_image: segment 3: paddr=0x0008c330 vaddr=0x3ffbe73c size=0x01760 (  5984) load[0m
[0;32mI (316) esp_image: segment 4: paddr=0x0008da98 vaddr=0x40080000 size=0x00404 (  1028) load[0m
[0;32mI (320) esp_image: segment 5: paddr=0x0008dea4 vaddr=0x40080404 size=0x0d398 ( 54168) load[0m
[0;32mI (359) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (360) boot: Disabling RNG early entropy source...[0m
[0;32mI (360) cpu_start: Pro cpu up.[0m
[0;32mI (364) cpu_start: Application information:[0m
[0;32mI (368) cpu_start: Project name:     ESP32_LyraT_Player[0m
[0;32mI (374) cpu_start: App version:      df4bbd1-dirty[0m
[0;32mI (380) cpu_start: Compile time:     Dec  2 2020 00:08:08[0m
[0;32mI (386) cpu_start: ELF file SHA256:  34b740d8e24b74f6...[0m
[0;32mI (392) cpu_start: ESP-IDF:          v4.1-dirty[0m
[0;32mI (397) cpu_start: Starting app cpu, entry point is 0x40081480[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (408) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (414) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM[0m
[0;32mI (421) heap_init: At 3FFC0880 len 0001F780 (125 KiB): DRAM[0m
[0;32mI (427) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (433) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (440) heap_init: At 4008D79C len 00012864 (74 KiB): IRAM[0m
[0;32mI (446) cpu_start: Pro cpu start user code[0m
[0;32mI (464) spi_flash: detected chip: gd[0m
[0;32mI (465) spi_flash: flash io: dio[0m
[0;32mI (465) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (472) BT_NETWORK_PLAYER: [1.0] Initialize peripherals management[0m
[0;32mI (482) BT_NETWORK_PLAYER: [1.1] Initialize and start peripherals[0m
[0;33mW (502) PERIPH_TOUCH: _touch_init[0m
[0;32mI (982) BT_NETWORK_PLAYER: [1.2] Set up a sdcard playlist and scan sdcard music save to it[0m
[0;32mI (1192) BT_NETWORK_PLAYER: [ 2 ] Start codec chip[0m
[0;31mE (1192) gpio: gpio_install_isr_service(438): GPIO isr service already installed[0m
[0;32mI (1222) BT_NETWORK_PLAYER: [ 3 ] Create and start input key service[0m
[0;32mI (1222) BT_NETWORK_PLAYER: [4.0] Create audio pipeline for playback[0m
[0;32mI (1232) BT_NETWORK_PLAYER: [4.1] Create i2s stream to write data to codec chip[0m
[0;32mI (1272) BT_NETWORK_PLAYER: [4.2] Create mp3 decoder to decode mp3 file[0m
[0;32mI (1272) BT_NETWORK_PLAYER: [4.3] Create resample filter[0m
[0;32mI (1272) BT_NETWORK_PLAYER: [4.4] Create fatfs stream to read data from sdcard[0m
[0;32mI (1282) BT_NETWORK_PLAYER: [4.5] Register all elements to audio pipeline[0m
[0;32mI (1292) BT_NETWORK_PLAYER: [4.6] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->resample-->i2s_stream-->[codec_chip][0m
[0;32mI (1312) BT_NETWORK_PLAYER: [5.0] Set up  event listener[0m
[0;32mI (1312) BT_NETWORK_PLAYER: [5.1] Listen for all pipeline events[0m
[0;33mW (1322) BT_NETWORK_PLAYER: [ 6 ] Press the keys to control music player:[0m
[0;33mW (1322) BT_NETWORK_PLAYER:       [Play] to start, pause and resume, [Set] next song.[0m
[0;33mW (1332) BT_NETWORK_PLAYER:       [Vol-] or [Vol+] to adjust volume.[0m
[0;32mI (8602) BT_NETWORK_PLAYER: [ * ] input key id is 3[0m
[0;32mI (8602) BT_NETWORK_PLAYER: [ * ] [Play] input key event[0m
[0;32mI (8602) BT_NETWORK_PLAYER: [ * ] Starting audio pipeline[0m
[0;32mI (9012) BT_NETWORK_PLAYER: [ * ] Received music info from mp3 decoder, sample_rates=44100, bits=16, ch=2[0m
[0;32mI (20902) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (20902) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (20902) BT_NETWORK_PLAYER: [ * ] Volume set to 40 %[0m
[0;32mI (23762) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (23762) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (23762) BT_NETWORK_PLAYER: [ * ] Volume set to 49 %[0m
[0;32mI (24952) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (24952) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (24952) BT_NETWORK_PLAYER: [ * ] Volume set to 58 %[0m
[0;32mI (26912) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (26912) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (26912) BT_NETWORK_PLAYER: [ * ] Volume set to 67 %[0m
[0;32mI (29452) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (29452) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (29452) BT_NETWORK_PLAYER: [ * ] Volume set to 76 %[0m
[0;32mI (32602) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (32602) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (32602) BT_NETWORK_PLAYER: [ * ] Volume set to 85 %[0m
[0;32mI (34702) BT_NETWORK_PLAYER: [ * ] input key id is 6[0m
[0;32mI (34702) BT_NETWORK_PLAYER: [ * ] [Vol+] input key event[0m
[0;32mI (34702) BT_NETWORK_PLAYER: [ * ] Volume set to 94 %[0m
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7008
ho 0 tail 12 room 4
load:0x40078000,len:13072
load:0x40080400,len:3896
entry 0x40080688
[0;32mI (31) boot: ESP-IDF v4.1-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 23:56:28[0m
```

jason-mao, 2020-12-02:
@KISSMonX Max value need more power, exceeds USB power supply capacity, so power reset.

closed 2020-12-02 by jason-mao

KISSMonX, 2020-12-02:
> @KISSMonX Max value need more power, exceeds USB power supply capacity, so power reset.

thanks。