espressif/esp-adf #527: error sending wav file to server

rjam-xav, 2020-12-02:
Hello,

I am reading the wav file in chunks and sending it to the azure server using  esp_http_client_write().
The write is sccessful when I don't send the data in chunks, which doesnt suit me because the file size is rather large and I need to transmit around 1mb of data to the server. After around 7kb of data I get the following error:

> I (18608) HTTP_CLIENT: Opened HTTP connection: ESP_OK
> I (18611) HTTP_CLIENT: bytes read 1096
> I (18613) HTTP_CLIENT:  WROTE DATA
> I (18615) HTTP_CLIENT: bytes read 1096
> I (18618) HTTP_CLIENT:  WROTE DATA
> I (18622) HTTP_CLIENT: bytes read 1096
> I (18627) HTTP_CLIENT:  WROTE DATA
> I (18631) HTTP_CLIENT: bytes read 1096
> I (18732) HTTP_CLIENT:  WROTE DATA
> I (18734) HTTP_CLIENT: bytes read 1096
> I (18737) HTTP_CLIENT:  WROTE DATA
> I (18739) HTTP_CLIENT: bytes read 1096
> I (18741) HTTP_CLIENT:  WROTE DATA
> I (18744) HTTP_CLIENT: bytes read 1096
> E (19043) esp-tls: write error :-80:
> E (19043) TRANS_SSL: esp_tls_conn_write error, errno=Connection reset by peer

My code for reading the file and writing it is as follows:

>   esp_err_t err;
>    // int write_len = 4096; 
>     if ((err = esp_http_client_open(client, 831532)) != ESP_OK) { //opens the stream and writes headers
>         ESP_LOGE(PTAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
>         esp_periph_set_destroy(set);
>         fclose(file);
>         free(buffer);
>         return 0;
>     }
>     ESP_LOGI(PTAG, "Opened HTTP connection: %s", esp_err_to_name(err));
> 
>     
> 
>     while (!feof(file))
>     {
>         char* buf = calloc(1, MAX_HTTP_OUTPUT_BUFFER + 1);
>         int bytesRead = fread(buf, 1, MAX_HTTP_OUTPUT_BUFFER, file);
>         if (ferror(file))
>         {
>             ESP_LOGW(PTAG, "Failed to read");
>             fclose(file);
>             esp_periph_set_destroy(set);
>             return 0;
>         }
>         ESP_LOGI(PTAG, "bytes read %d", bytesRead);
>         ESP_LOG_BUFFER_CHAR(PTAG, buf, bytesRead);
>         if (esp_http_client_write(client, buf, bytesRead) <= 0) {
>             ESP_LOGI(PTAG, "ERROR WRITING DATA");
>             fclose(file);
>             esp_periph_set_destroy(set);
>             return 0;
>         }
>         ESP_LOGI(PTAG, " WROTE DATA");
>         free(buf);
>     }
>     ESP_LOGW(PTAG, "FILE READ");
>     
>     fclose(file);

What could be causing this issue?

jason-mao, 2020-12-07:
@rjam-xav Which IDF version you used? Did you test more with other AP? `errno=Connection reset by peer`  look like socket has been closed by server.

rjam-xav, 2020-12-08:
Yes the problem was with asking the server to send chunked data. I was able to bypass this problem by opening the socket with the full file size that I wanted to write in place of the write_len argument and then writing in chunks. Thank you

closed 2020-12-08 by rjam-xav