espressif/esp-adf #528: Algorithm stream in VOIP example issue

Patsjemoe, 2020-12-02:
Hi,
I added the algorithm stream to the VOIP example. Replaced the recorder_pipeline_open() and player_pipeline_open(); by one part taking both recorder and player as in the algorithm example.
As it is (see below), no AEC is executed. WHen I leave the rsp out of the recorder part, and have put rec_sample rate and ref_sample rate at 16000, rec_ch=1, ref_ch - 2, I get a grainy sound on my android phone. My voice comes OK out of my speaker of my esp32.
Please have a look if you see  some mistakes : 
	ESP_LOGI(TAG, " **************recorder begin***************");
	
	ESP_LOGI(TAG, " create audio pipeline recorder for recording");
    audio_element_handle_t i2s_stream_reader;
    audio_pipeline_cfg_t pipeline_rd_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    recorder = audio_pipeline_init(&pipeline_rd_cfg);
	mem_assert(recorder);
    AUDIO_NULL_CHECK(TAG, recorder, return ESP_FAIL);

	ESP_LOGI(TAG, " create I2S stream to read from codec ");
    i2s_stream_cfg_t i2s_rd_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_rd_cfg.type = AUDIO_STREAM_READER;
    i2s_rd_cfg.uninstall_drv = false;
#if defined CONFIG_ESP_LYRAT_MINI_V1_1_BOARD
    i2s_rd_cfg.i2s_port = 1;
#endif
    i2s_stream_reader = i2s_stream_init(&i2s_rd_cfg);
    audio_element_info_t i2s_rd_info = {0};
    audio_element_getinfo(i2s_stream_reader, &i2s_rd_info);
    i2s_rd_info.bits = I2S_BITS;
    i2s_rd_info.channels = I2S_CHANNELS;
    i2s_rd_info.sample_rates = I2S_SAMPLE_RATE;
    audio_element_setinfo(i2s_stream_reader, &i2s_rd_info);

	ESP_LOGI(TAG, "[10.2] Create algo stream ");
    algorithm_stream_cfg_t algo_config = ALGORITHM_STREAM_CFG_DEFAULT();
    algo_config.input_type = ALGORITHM_STREAM_INPUT_TYPE2;
    audio_element_handle_t element_algo = algo_stream_init(&algo_config);

	ESP_LOGI(TAG, "[] Create rsp filter ");
    rsp_filter_cfg_t rsp_rd_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_rd_cfg.src_rate = I2S_SAMPLE_RATE;
    rsp_rd_cfg.src_ch = I2S_CHANNELS;
    rsp_rd_cfg.dest_rate = CODEC_SAMPLE_RATE;
    rsp_rd_cfg.dest_ch = CODEC_CHANNELS;
    rsp_rd_cfg.complexity = 5;
    audio_element_handle_t filter_rd = rsp_filter_init(&rsp_rd_cfg);

	ESP_LOGI(TAG, " Create sip encoder ");
    g711_encoder_cfg_t g711_rd_cfg = DEFAULT_G711_ENCODER_CONFIG();
    audio_element_handle_t sip_encoder = g711_encoder_init(&g711_rd_cfg);

		ESP_LOGI(TAG, "Create raw stream ");
    raw_stream_cfg_t raw_rd_cfg = RAW_STREAM_CFG_DEFAULT();
    raw_rd_cfg.type = AUDIO_STREAM_READER;
    raw_read = raw_stream_init(&raw_rd_cfg);
    audio_element_set_output_timeout(raw_read, portMAX_DELAY);

    audio_pipeline_register(recorder, i2s_stream_reader, "i2s");
	audio_pipeline_register(recorder, filter_rd, "filterrd");
	audio_pipeline_register(recorder, element_algo, "algo");
    audio_pipeline_register(recorder, sip_encoder, "sip_enc");
    audio_pipeline_register(recorder, raw_read, "raw");
    const char *link_rec[5] = {"i2s", "filterrd", "algo", "sip_enc", "raw"};
    audio_pipeline_link(recorder, &link_rec[0], 5);
    audio_pipeline_run(recorder);
    ESP_LOGI(TAG, " SIP recorder has been created");

	ESP_LOGI(TAG, " **************player begin***************");
	
	ESP_LOGI(TAG, " create audio pipeline player");
    audio_element_handle_t i2s_stream_writer;
    audio_pipeline_cfg_t pipeline_pl_cfg = DEFAULT_AUDIO_PIPELINE_CONFIG();
    player = audio_pipeline_init(&pipeline_pl_cfg);
	mem_assert(player);
    AUDIO_NULL_CHECK(TAG, player, return ESP_FAIL);

	ESP_LOGI(TAG, " create RAW stream");
    raw_stream_cfg_t raw__pl_cfg = RAW_STREAM_CFG_DEFAULT();
    raw__pl_cfg.type = AUDIO_STREAM_WRITER;
    raw_write = raw_stream_init(&raw__pl_cfg);

	ESP_LOGI(TAG, " create sip decoder");
    g711_decoder_cfg_t g711_pl_cfg = DEFAULT_G711_DECODER_CONFIG();
    audio_element_handle_t sip_decoder = g711_decoder_init(&g711_pl_cfg);

	ESP_LOGI(TAG, " create rsp filter player");
    rsp_filter_cfg_t rsp_pl_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_pl_cfg.src_rate = CODEC_SAMPLE_RATE;
    rsp_pl_cfg.src_ch = CODEC_CHANNELS;
    rsp_pl_cfg.dest_rate = I2S_SAMPLE_RATE;
    rsp_pl_cfg.dest_ch = I2S_CHANNELS;
    rsp_pl_cfg.complexity = 5;
    audio_element_handle_t filter_pl = rsp_filter_init(&rsp_pl_cfg);

	ESP_LOGI(TAG, " create I2S ");
    i2s_stream_cfg_t i2s_pl_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_pl_cfg.type = AUDIO_STREAM_WRITER;
    i2s_pl_cfg.uninstall_drv = false;
    i2s_stream_writer = i2s_stream_init(&i2s_pl_cfg);
    audio_element_info_t i2s_pl_info = {0};
    audio_element_getinfo(i2s_stream_writer, &i2s_pl_info);
    i2s_pl_info.bits = I2S_BITS;
    i2s_pl_info.channels = I2S_CHANNELS;
    i2s_pl_info.sample_rates = I2S_SAMPLE_RATE;
    audio_element_setinfo(i2s_stream_writer, &i2s_pl_info);

	ESP_LOGI(TAG, "link together");
    audio_pipeline_register(player, raw_write, "raw");
    audio_pipeline_register(player, sip_decoder, "sip_dec");
    audio_pipeline_register(player, filter_pl, "filterpl");
    audio_pipeline_register(player, i2s_stream_writer, "i2s");
    const char *link_pl[4] = {"raw", "sip_dec", "filterpl", "i2s"};
    audio_pipeline_link(player, &link_pl[0], 4);
    audio_pipeline_run(player);
    ESP_LOGI(TAG, "Speaker has been created");
	
	ESP_LOGI(TAG, " create ringbuf ");
	ringbuf_handle_t input_rb = algo_stream_get_multi_input_rb(element_algo);
	audio_element_set_multi_output_ringbuf(sip_encoder, input_rb, 0);
	algo_stream_set_reference_rate(element_algo, ref_ch, ref_sample_rate);
	algo_stream_set_record_rate(element_algo, rec_ch, rec_sample_rate);
    return ESP_OK;
}

Patsjemoe, 2020-12-07:
Hi,
Any suggestion? Voip without AEC is not usable.
Regards
Ludo

Patsjemoe, 2020-12-10:
Please feedback/ suggestions

Patsjemoe, 2020-12-14:
Hi,
please confirm if I can expect feedback or not ?
regards
Ludo

jason-mao, 2020-12-15:
Hi Ludo, Sorry for late. The `algo_stream_init ` already integrate the resample module, https://github.com/espressif/esp-adf/blob/e4a1c52dab7a951ec4894d8aa082e831a3fcd608/components/audio_stream/algorithm_stream.c#L687 and https://github.com/espressif/esp-adf/blob/e4a1c52dab7a951ec4894d8aa082e831a3fcd608/components/audio_stream/algorithm_stream.c#L657 can be modify the two source rate,channel for resample parameters. So as your code, there is not need `filterrd ` before the algorithm.

Patsjemoe, 2020-12-15:
@jason-mao , thank you for your feedback.
As AEC does not work when taking SIP_decoder  or filter_pl  as multi output,I am now trying to use i2s_stream_writer as multi output.
audio_element_set_multi_output_ringbuf(i2s_stream_writer, input_rb, 0);
algo_stream_set_reference_rate(element_algo, 2, 48000); 
algo_stream_set_record_rate(element_algo, 1, 48000)
I have 2 48K I2S inputs to Algo_stream, but the output is 16K. As the SIP encoder is at 8000 samples, I tried to change 
#define ALGORITHM_STREAM_DEFAULT_SAMPLE_RATE_HZ   16000 ....to 8000, but then have an error 
I (4953) VOIP_EXAMPLE: [10.2] Create algo stream
E (4953) LI_RESAMPLE: in encode mode, max_indata_bytes(4096) is too small.
In algorithm_stream.c, I found that I am at the limit due to having 2x 48Kin : 
The max input data maybe 48K 2ch --> 16k 1ch; so max_data = AEC_FRAME_BYTES * 6
How can I solve this ? (I can put an rsp behind from 16K to 8K, but this still does not make AEC function.
regards
Ludo

jason-mao, 2020-12-18:
@Patsjemoe We plan to modify the VOIP example to support algorithm_stream.

Patsjemoe, 2020-12-18:
Hi,
Thank you.
 Looking forward to it.
Regards
Ludo

Patsjemoe, 2020-12-29:
@jason-mao : hi,  any progress/schedule?

Patsjemoe, 2021-01-09:
> @Patsjemoe We plan to modify the VOIP example to support algorithm_stream.

@jason-mao : it would be nice if you can confirm if the project  started already?

jason-mao, 2021-01-11:
@Patsjemoe Yes, we are working on this.The AEC performance is not good due to the VOIP specific features . We are trying to optimize it. But it would be take a lot of  time.

Patsjemoe, 2021-01-11:
Hi, Jason-mao
Thank you for confirming that  work is ongoing.
If you have an intermediate solution that has some improvement, I will be
happy to test it. You can mail the manual file adaptations before an
official esp-adf update.
Thank you in advance for the work.
Regards
Ludo

Patsjemoe, 2021-01-25:
@jason-mao : any progress?
Thanks in advance
Ludo

Patsjemoe, 2021-02-02:
@jason-mao : is it possible to mute the microphone when voice is coming in ?

jason-mao, 2021-02-03:
@Patsjemoe  We update the VOIP example  on 0b715aa5a01c6e999e5c98730e324325f6770c1d. Please check. And there is thing need to note that we used the LyraT-mini board.

Patsjemoe, 2021-02-03:
Hi @jason-mao, thank you already for your help.
I do not have mini, only lyrat and audiokit. Any good reason to use mini?  
Will look into the details....

jason-mao, 2021-02-03:
Pelase see https://github.com/espressif/esp-adf/issues/570#issuecomment-772272634

Patsjemoe, 2021-02-16:
> @Patsjemoe We update the VOIP example on [0b715aa](https://github.com/espressif/esp-adf/commit/0b715aa5a01c6e999e5c98730e324325f6770c1d). Please check. And there is thing need to note that we used the LyraT-mini board.

@jason-mao : which IDF and ADF version did you use ?
I have issues reported to michael, but do not seem to get feedback.
regards
ludo

jason-mao, 2021-02-18:
@Patsjemoe Hi Ludo, sorry for late. We are just back to work due to the festival holidays. We will reply message on your issue.

closed 2021-03-03 by Patsjemoe