espressif/esp-adf #529: esp-adf memory corrupt

huybk213, 2020-12-04:
Hi all,

I tested example HLS, after running about 7h audio streaming, assertion occurs.
Below is my log, could you please take a look and give me suggesstion?

1. ESP_ADF version 3.3
2. ESP-IDF release/4.2
3. Example HLS
4. URL https://strm.voh.com.vn/radio/channel3/chunklist_w1487815672.m3u8

Log

> W (23560599) HLS_PLAYLIST: URI exist
W (23560600) HLS_PLAYLIST: URI exist
W (23570624) HTTP_STREAM: No more data,errno:0, total_bytes:lu, rlen = 121868
W (23570688) HLS_PLAYLIST: URI exist
W (23570688) HLS_PLAYLIST: URI exist
W (23580746) HTTP_STREAM: No more data,errno:0, total_bytes:lu, rlen = 122144
W (23580834) HLS_PLAYLIST: URI exist
W (23580835) HLS_PLAYLIST: URI exist
W (23590870) HTTP_STREAM: No more data,errno:0, total_bytes:lu, rlen = 122075
E (23590875) HTTP_STREAM: Invalid HTTP stream, status code = -1
E (23590875) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN,-1
W (23590879) AUDIO_ELEMENT: [http] AEL_MSG_CMD_ERROR,6
W (23592596) AUDIO_ELEMENT: IN-[aac] AEL_IO_ABORT
CORRUPT HEAP: multi_heap.c:172 detected at 0x3fdaf210
abort() was called at PC 0x4002fb0b on core 0
0x4002fb0b: multi_heap_assert at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) assert_valid_block at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/multi_heap.c:172
Backtrace:0x4002c6e3:0x3ffee6f0 0x4002cdb9:0x3ffee710 0x40030412:0x3ffee730 0x4002fb0b:0x3ffee7a0 0x4002fde2:0x3ffee7c0 0x400254d2:0x3ffee7e0 0x4003043d:0x3ffee800 0x400a6d45:0x3ffee820 0x400a6c93:0x3ffee840 0x40094947:0x3ffee860 0x4008fdd3:0x3ffee880 0x4008fecd:0x3ffee8a0 0x4012ac41:0x3ffee8c0 0x4008a944:0x3ffee8e0 0x4008b927:0x3ffee910 0x4008ac97:0x3ffee950 0x4002cdc1:0x3ffee980
0x4002c6e3: panic_abort at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/esp_system/panic.c:330
0x4002cdb9: esp_system_abort at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/esp_system/system_api.c:106
0x40030412: abort at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/newlib/abort.c:46
0x4002fb0b: multi_heap_assert at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/multi_heap_platform.h:67
 (inlined by) assert_valid_block at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/multi_heap.c:172
0x4002fde2: multi_heap_free_impl at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/multi_heap.c:539
0x400254d2: heap_caps_free at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/heap/heap_caps.c:305
0x4003043d: free at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/newlib/heap.c:47
0x400a6d45: audio_free at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/components/audio_sal/audio_mem.c:52
0x400a6c93: codec_memory_free at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/codec-utils/codec_mem.c:139
0x40094947: PVMP4AudioDecoderDeInit at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_codec/esp-aac/src/pvmp4audiodecoderinitlibrary.c:362
0x4008fdd3: aac_decoder_close at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/aac_decoder.c:652
0x4008fecd: _aac_decoder_close at /home/houhaiyan/esp-audio-adf-projects/esp-adf-libs-source/esp_processing/esp-wrapper/aac_decoder.c:958
0x4012ac41: audio_element_process_deinit at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/components/audio_pipeline/audio_element.c:199 (discriminator 15)
0x4008a944: audio_element_on_cmd at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/components/audio_pipeline/audio_element.c:241
0x4008b927: audio_event_iface_waiting_cmd_msg at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/components/audio_pipeline/audio_event_iface.c:246
0x4008ac97: audio_element_task at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/components/audio_pipeline/audio_element.c:453
0x4002cdc1: vPortTaskWrapper at /media/huytv/Work/tmp/audio_toy/build/../ipa_esp_adf/esp-idf/components/freertos/xtensa/port.c:143
ELF file SHA256: 2ea2904acc10ef7d
CPU halted.

**- My example code**

`
```
void app_main(void)
{
    ESP_LOGI(TAG, "Build %s %s\r\n", __DATE__, __TIME__);
    esp_err_t err = nvs_flash_init();
#if ETHERNET_ENABLE
    ethernet_start();
#endif
#if PPPOS_ENABLE
    pppos_start();
#endif
    // vTaskDelay(5000);
    if (err == ESP_ERR_NVS_NO_FREE_PAGES) {
        // NVS partition was truncated and needs to be erased
        // Retry nvs_flash_init
        ESP_ERROR_CHECK(nvs_flash_erase());
[... 169 lines cut ...]
    audio_pipeline_wait_for_stop(pipeline);
    audio_pipeline_terminate(pipeline);

    /* Terminate the pipeline before removing the listener */
    audio_pipeline_unregister(pipeline, http_stream_reader);
    audio_pipeline_unregister(pipeline, i2s_stream_writer);
    audio_pipeline_unregister(pipeline, mp3_decoder);

    audio_pipeline_remove_listener(pipeline);

    /* Stop all peripherals before removing the listener */
    esp_periph_set_stop_all(set);
    audio_event_iface_remove_listener(esp_periph_set_get_event_iface(set), evt);

    /* Make sure audio_pipeline_remove_listener & audio_event_iface_remove_listener are called before destroying event_iface */
    audio_event_iface_destroy(evt);

    /* Release all resources */
    audio_pipeline_deinit(pipeline);
    audio_element_deinit(http_stream_reader);
    audio_element_deinit(i2s_stream_writer);
    audio_element_deinit(mp3_decoder);
    esp_periph_set_destroy(set);
}
````

jason-mao, 2020-12-07:
@huybk213 Did you use the https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v3.3_esp_http_client.patch  and https://github.com/espressif/esp-adf/blob/master/idf_patches/adf_http_stream.patch for your project?

huybk213, 2020-12-07:
> @huybk213 Did you use the https://github.com/espressif/esp-adf/blob/master/idf_patches/idf_v3.3_esp_http_client.patch and https://github.com/espressif/esp-adf/blob/master/idf_patches/adf_http_stream.patch for your project?

Hi @jason-mao
I didnot.
I will check it out.

jason-mao, 2020-12-15:
@huybk213 How about this issue?

huybk213, 2020-12-15:
hi @jason-mao 
After 36 hours streaming, this issues seem fixed.
Thanks.

closed 2020-12-15 by huybk213