espressif/esp-adf #531: navigation in m4a song with high accurate #519

mcSimoffa, 2020-12-04:
Jason, thanks for you help, but I need a little bigger.
I need to navigate with accurate about 22ms (one m4a block = 1024 samle. sample rate = 44100). esp_audio_seek() have an  accurate about 1 sec. So I develope such alogritm:
```
//here music is sound
audio_element_pause(aac_decoder);
rb_reset`(ringbuf);
m4a_seek(position, &bytePos, );  /* this is my function. It parse atoms 'stts' and 'stsz' 
and calculate the position in 'mdat' atom. 
This position I use in callback: read_cb(audio_element_handle_t el, char *buf, int len, TickType_t wait_time, void *ctx) function 
for fseek() function. */
audio_element_set_byte_pos(aac_decoder, bytePos);
audio_element_resume(aac_decoder,0,100); //decoder wake up and read the block which need for me!
```
Such I feed the decoder a need block. It work, the music jump! It have positioning with absolutly accurate. 
But there is two bad point:
1. I cath a error 'M4A_PARSER: can not seach the sample ID , line 1048'  after jump. it doesn't matter because it continue work, the musiccan play and jump many times.
2. But when time passes equal to the duration of my song, then the decoder will stop. It can stop in the middle of the song sounded (I mooved the song back several times). It stops. Obviously, I must reset the counter of decoded samples. 
But how can I do this?
Even if I use a standard seek function a**udio_element_seek()**- my music also stopped before reaching the end of the track. With message AAC_DECODER: Closed by [2]. I want to stop music only at end of track, when callback:  **read_cb()** back EOF.
msg source type 131072   msg source 0x3f810e04   msg cmd 8   msg data 0xf
0x3f810e04  - it's AAC decoder handle
The aac decoder source file have *.a shape. I don't see code, I don't know how it work in detail. If it's easy say me please: how to assign a sample counter to new value

jason-mao, 2020-12-15:
@mcSimoffa  What is your scenario need so high accurate? The high level api round up to an integer second.

mcSimoffa, 2020-12-15:
I develop device for syncronus play the music on several device. I would be agree with integer seconds positioning, but even integer second isn't integer. For example: I set 5 second, but music positioning to 5,3 second. It working approximately with +-0.5 sec
If you have a header files for library libesp_aac.a and libesp_processing.a, I would to be thanks.