espressif/esp-adf #537: ALGORITHM_STREAM_INPUT_TYPE2 error in description ?

Patsjemoe, 2020-12-11:
The example shows in .h file : 
ALGORITHM_STREAM_INPUT_TYPE2 = 2, /*!< Type 2 read in record signal from I2S and when data be written, the data should be copy as a reference signal and input to the algorithm element by using multiple input buffer. */
Shouldn't that be :  
copy as a **recording** signal and input to the algorithm element by using multiple input buffer. */

Patsjemoe, 2020-12-17:
Hi,
I expected this to be easy to answer.... Wonder why not? 
Regards
Ludo

withdra, 2020-12-17:
Hi @Patsjemoe, sorry to forget to reply to your comment.
Yeah, you are right, the comments in this part are not very accurate. We will modify that soon, please update the code in one or two days. Thanks for your report !

jason-mao, 2021-02-23:
Fixed by https://github.com/espressif/esp-adf/commit/61c91bdd4fd2a45d2a9c79f7c18df3a393330cde

closed 2021-02-23 by jason-mao